mpromonet/webrtc-streamer #361: Application Flow

petryx, 2020-11-12:
Hi Michael,

Firstable your project is awesome. Congratulations for your code.

I'm trying understand your code, would be possible you can describe the application flow of point of view  of IP Camera with rtsp until video appear on browser.

Thanks in advance Michael

mpromonet, 2020-11-13:
Hi petryx,

The flow is very simple, it makes a very simple protocol conversion, the RTSP Sink frame is decoded (using WebRTC decoder) and post to WebRTC Source.

Best Regards,
Michel.

petryx, 2020-11-13:
Thanks Michel.

Why need develop a very simple version, with the minimal necessary.

> Hi petryx,
> 
> The flow is very simple, it makes a very simple protocol conversion, the RTSP Sink frame is decoded (using WebRTC decoder) and post to WebRTC Source.
> 
> Best Regards,
> Michel.

closed 2020-12-09 by mpromonet