mpromonet/webrtc-streamer #362: Error To build

petryx, 2020-11-13:
Hi Michael,

Need some help to build the app. I'm getting the following error
`webrtc-streamer/inc/VideoDecoder.h:63:33: error: ‘class webrtc::EncodedImage’ has no member named ‘_completeFrame’
   63 |                     input_image._completeFrame = true;
      |                                 ^~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/webrtc-streamer.dir/build.make:76: CMakeFiles/webrtc-streamer.dir/src/PeerConnectionManager.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:127: CMakeFiles/webrtc-streamer.dir/all] Error 2
make: *** [Makefile:152: all] Error 2
`

petryx, 2020-11-13:
Update:

The variable **_completeFrame** was removed.

https://bugs.chromium.org/p/webrtc/issues/detail?id=9378#c91

mpromonet, 2020-11-13:
Hi petryx,

Did you rtied to remove the line ?

Best Regards,
Michel.

closed 2020-11-13 by mpromonet

petryx, 2020-11-13:
Yes Michel, I removed the line and work like a charm.

Thanks

> Hi petryx,
> 
> This line may be obsolete with new code. Did you try to remove the line ?
> 
> Best Regards,
> Michel.

commit 9458b5d3 (2020-11-13): fix #362: build with latest WebRTC