mpromonet/webrtc-streamer #363: RTP / UDP H264 stream compatibility

lukelu0520, 2020-11-18:
Dear webrtc-streamer friends,

Recently, we are receiving a RTP / UDP H264 video stream from a device, and we are considering re-stream it via webrtc-streamer with our server.

We were wondering if any suggested ways to make it. Do we need to convert the RTP / UDP H264 stream to a RTSP stream ? Or webrtc-streamer has an internal method to capture RTP / UDP H264 and make it available. 

Hope to hear your expertise.

Best,
-- Luke

mpromonet, 2020-11-18:
Dear Luke,

As webrtc-streamer use live555helper to get RTSP stream that also support RTP stream, it should be easy to add RTP support to the consumer.

Best Regards,
Michel.

lukelu0520, 2020-11-18:
@mpromonet , sounds nice, thanks for your prompt reply! So you mean just simply apply url _rtp://ip_address:port_ ? 
Kindly let us know if any reference (or readme) in order to implement it.

Best,
-- Luke

lukelu0520, 2020-11-20:
@mpromonet, Update: we tested many configurations like:  
`./webrtc-streamer rtp://[bind_host][:port]@[remote_host[:port]]` or
`./webrtc-streamer udp://[bind_host][:port]@[remote_host[:port]]`

But we can not get the stream so far. Hope to hear your expertise.
Best,
-- Luke

mpromonet, 2020-11-20:
Hi Luke,

It is needed to modify the code, in the actual situation this is not possible. I was explaining a way to easily adding this feature.

Best Regards,
Michel.

closed 2022-05-13 by mpromonet