mpromonet/webrtc-streamer #365: Add Remote WebRTC server capability

yeshivsher, 2020-11-23:
I'm using the Docker option of this project installed locally on a computer that connected to multiple cameras.

I want to stream these cameras to the internet.

how can I do it?

what I think is the answer is to create a server that the client goes to ask the URL of this project but I don't know how to connect this project to a WebRTC server.

mpromonet, 2020-11-24:
Hi,

WebRTC could be streamed over internet, you may check ICE candidate enumeration.

Best Regards,
Michel.

yeshivsher, 2020-11-26:
What do you mean by " check ICE candidate enumeration"
can you add more details?

gtntdev, 2020-12-04:
You need at least one server public available, at least you have to install a Turn server on this instace. Then pass the credentials from the turn server as described by the README of this project to the webrtc Streamer instance either by config file or as cmd args. Ofcourse you could rent such a service but I guess it makes more fun doing it on your own especially with the coturn project. Only with a Turn server you are guarented to overcome all NATs and stuff like that.

yeshivsher, 2020-12-08:
Thanks!
Can you share a link with more info of how to do it?
I'm not familiar with the Turn server.

mpromonet, 2020-12-09:
Hi @yeshivsher 

For a first try, you can configure your NAT to forward http port of webrtc-streamer. In many case this should work.
Did you try that ?

Best Regards,
Michel

yeshivsher, 2020-12-10:
Are you talking about port forwarding?
How can I do "NAT forward HTTP"?

mpromonet, 2020-12-10:
Maybe http://letmegooglethat.com/?q=nat+forward+port

yeshivsher, 2020-12-10:
Yeah I know lol...
But I did succeed in publishing the app the problem is when the consumer (the client from a different network) tries to get the streaming here the webrtc cant finds his IP - maybe I need to install Turn server.
The question of how for this project.

gtntdev, 2020-12-10:
For getting started you just could `sudo apt-get install coturn` modify config file an test your settings [here](https://webrtc.github.io/samples/src/content/peerconnection/trickle-ice/). You need to get a Turn Candidate entry with prefix "relay". There're some docker images available too, don't forget to use the `--net=host` option.

mpromonet, 2020-12-11:
Hi,
From the very short description, it is difficult to imagine what could be your problem. It could be ICE configutation, or network configuration, or something else...
If you like to try to use TURN, in case STUN is not sufficient, you could try the embeded TURN server, maybe starting it with 
```
./webrtc-streamer -s- -T0.0.0.0:3478 -tturn:turn@$(curl -s ifconfig.me):3478
```
This need to NAT the TURN port (3478)
Best Regards,
Michel.

closed 2021-01-01 by mpromonet