mpromonet/webrtc-streamer #366: Docker image with v4l2 and ALSA capture

gdhgdhgdh, 2020-11-27:
Hello, I'm able to see my own webcam, but not hear any audio. I spotted a few other GitHub issues which have helped me get closed, but there's still silence:

`docker run --rm --device /dev/video0 --device /dev/snd -p 8083:8000 -it mpromonet/webrtc-streamer  -vvv`

When I curl `/api/getAudioDeviceList` then I get the only capture device installed in the machine - perfect!

```
uri:/api/getAudioDeviceList
answer:[
	"default",
	"HDA Intel PCH, ALC293 Analog-Default Audio Device",
	"HDA Intel PCH, ALC293 Analog-Default Audio Device",
	"HDA Intel PCH, ALC293 Analog-Front speakers",
	"HDA Intel PCH, ALC293 Analog-Direct sample snooping device",
	"HDA Intel PCH, ALC293 Analog-Direct hardware device without any conversions",
	"HDA Intel PCH, ALC293 Analog-Hardware device with all software conversions"
]
```

If I visit http://localhost:8083/webrtcstreamer.html?video=Integrated%20Camera%3A%20Integrated%20C then I see my own ugly face :+1: :)

How do I add the audio? I've tried appending `&audio=default` to the URL but that's made no difference. What silly mistake have I made?

How do I actually tell the

mpromonet, 2020-12-06:
Hi @gdhgdhgdh 

At a glance this seems correct. Did you try without the docker layer ?

Best Regards,
Michel.

gdhgdhgdh, 2020-12-06:
Michel,

Yes, indeed. The binary release of 0.3.3 has the same behaviour. I even tried a selection of 0.2.x and 0.1.x binaries for good measure. None were able to generate an RTP audio stream with any actual audio content.

As a fudge I am using the RTSP facility of webrtc-streamer and an external process to make the ALSA capture available as an Opus stream. This works, but adds complexity and latency. I would love to have the native capture working!

Cheers
Gavin.

mpromonet, 2020-12-06:
Hi Gavin,

I made some tries and it doesnot seems to work anymore, maybe since a while....
In the logs it seems to says that audio device is busy. 
I restore the code that looks for audio+video device, it was my initial idea to make work my webcam with audio and video.

it needs some debugging time.

Best Regards,
Michel.

mpromonet, 2020-12-12:
Hi Gavin,

One part of the problem is the html that start "muted" because it is not allowed to autostart audio.
I made some modification to access video control and then give the opportunity to enable sound.

Best Regards,
Michel.

gdhgdhgdh, 2020-12-12:
Oh my, adding pulse support is an even more useful direction. For my use case, the audio source is a PulseAudio client, so this will mean I can remove the ALSA `snd-aloop` device entirely `\o/`

closed 2020-12-17 by mpromonet