mpromonet/webrtc-streamer #367: Segmentation fault on Ubuntu 20.04

PooyaEimandar, 2020-11-29:
**Segmentation fault on Ubuntu 20.04**

First of all thank you for contributing to the open source community.
I just setup this project on Ubuntu 20.04 and integrate it with the latest branch on webRTC.
Everything works fine except **api/hangUP** 
I faced with the following error:

```
uri:/api/hangup
[022:150][1067407] (PeerConnectionManager.cpp:744): Remove PeerConnection peerid:0.45061071241328254
[022:150][1067407] (PeerConnectionManager.cpp:761): hangUp stream is no more used 7601519624931878834
[022:150][1067407] (PeerConnectionManager.cpp:769): hangUp stream closed 7601519624931878834
Segmentation fault
```
I checked debug information and found out that  `peerConnection->RemoveStream(stream);` threw exception. it seems that **[RemoveStream](https://developer.mozilla.org/en-US/docs/Web/API/RTCPeerConnection/removeStream)** is deprecated in the latest version of webRTC, and we need to use RemoveTrack instead of it.

gferandelle, 2020-11-30:
I have the same problem with the last version using docker.

mpromonet, 2020-12-05:
Hi @gferandelle @PooyaEimandar 

I am not reproducing what you are describing, however I fixed a crash https://github.com/mpromonet/webrtc-streamer/commit/3ad66377d288017e4b9c2584e503689bc49393c3 that occurs closing a stream (this is not linked to usage of a deprecated api, but usage of a NULL pointer).   
Did you have still the crash on a version with this fix ?

Best Regards,
Michel.

PooyaEimandar, 2020-12-06:
@mpromonet 
Works like charm, thank you.

closed 2020-12-06 by mpromonet