mpromonet/webrtc-streamer #369: Issue with empty ICE server list

neamjr, 2020-12-11:
If I turn off STUN with the `-s-` flag, the `getIceServers` api call returns this:

```
{
	"iceServers" : null
}
```

This causes an error in the client. The correct response should be:

```
{
	"iceServers" : []
}
```

I think this can be fixed on this line, but I don't have a working build (using the Docker image):

https://github.com/mpromonet/webrtc-streamer/blob/da0ea2bc9023aeffbec481fd774420dcf464b422/src/PeerConnectionManager.cpp#L405

mpromonet, 2020-12-11:
Hi neamjr,

You are right, it should be something like `Json::Value urls(Json::arrayValue);`.
However if you have not any iceservers, I guess it will works only on LAN.

Best Regards,
Michel.

neamjr, 2020-12-11:
Hi Michel,

Yes, that is correct. My use case is on a LAN without Internet connectivity.

Thanks,
Michael

commit b54c6f46 (2020-12-11): fix #369: manage case without any ICE servers
closed 2020-12-11 by mpromonet