mpromonet/webrtc-streamer #370: Adding ALSA or PulseAudio capture to `Dockerfile.arm64`

gdhgdhgdh, 2020-12-12:
Hello,

The ARM64 Dockerfile works perfectly and builds a Docker image. I note that during the build, neither ALSA nor PulseAudio are found.

I tried simply adding `libasound2-dev libpulse-dev` to the `apt-get install` on line 12 (as happens in the main `Dockerfile`) but this made no difference. What am I doing wrong?

```
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/aarch64-linux-gnu-gcc
-- Check for working C compiler: /usr/bin/aarch64-linux-gnu-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/aarch64-linux-gnu-g++
-- Check for working CXX compiler: /usr/bin/aarch64-linux-gnu-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
WEBRTCBUILD = Release
WEBRTCROOT = /webrtc-streamer/../webrtc
WEBRTCDESKTOPCAPTURE= OFF
CMAKE_CXX_COMPILER_ID=GNU
-- Found Git: /usr/bin/git (found version "2.17.1") 
ALSA_FOUND = FALSE
PulseAudio_FOUND = 0
GTK_FOUND = 
```

For reference, building using the existing `Dockerfile` also works perfectly, and DOES include ALSA + PulseAudio support:
```
-- Detecting CXX compile features - done
WEBRTCBUILD = Release
WEBRTCROOT = /webrtc-streamer/../webrtc
WEBRTCDESKTOPCAPTURE= ON
CMAKE_CXX_COMPILER_ID=GNU
-- Found Git: /usr/bin/git (found version "2.17.1") 
ALSA_FOUND = TRUE
PulseAudio_FOUND = 1
GTK_FOUND = 1
Done. Made 914 targets from 240 files in 175ms

```

mpromonet, 2020-12-12:
Hi gdhgdhgdh,

Adding `apt get install libasound2-dev libpulse-dev` to Dockerfile.arm64 add x86-64 package, you needs of arm64 packages. These packages may be existing in the WebRTC sysroot for arm64, but the cmakelist.txt doesnot look into it.... something to improve...

Best Regards,
Michel.

gdhgdhgdh, 2020-12-12:
Ah that makes sense. I had a quick peek inside the sysroot mentioned in the build log output:

```
# curl -LO https://commondatastorage.googleapis.com/chrome-linux-sysroot/toolchain/ff4e6495cf3db79fc54addcb340f6f951885e4c1/debian_sid_arm64_sysroot.tar.xz
# tar tvf debian_sid_arm64_sysroot.tar.xz  | egrep 'usr.lib.*libasound2|usr.lib.*libpulse'
drwxr-xr-x thomasanderson/primarygroup        0 2019-11-05 15:50 ./usr/include/alsa/
-rw-r--r-- thomasanderson/primarygroup    60758 2019-11-05 15:50 ./usr/include/alsa/pcm.h
-rw-r--r-- thomasanderson/primarygroup    16699 2019-11-05 15:50 ./usr/include/alsa/use-case.h
-rw-r--r-- thomasanderson/primarygroup     2760 2019-11-05 15:50 ./usr/include/alsa/output.h
-rw-r--r-- thomasanderson/primarygroup     8003 2019-11-05 15:50 ./usr/include/alsa/pcm_ioplug.h
-rw-r--r-- thomasanderson/primarygroup     2592 2019-11-05 15:50 ./usr/include/alsa/input.h
-rw-r--r-- thomasanderson/primarygroup     1853 2019-11-05 15:50 ./usr/include/alsa/asoundlib.h
-rw-r--r-- thomasanderson/primarygroup      497 2019-11-05 15:50 ./usr/include/alsa/version.h
-rw-r--r-- thomasanderson/primarygroup     8793 2019-11-05 15:50 ./usr/include/alsa/control_external.h
-rw-r--r-- thomasanderson/primarygroup     7062 2019-11-05 15:50 ./usr/include/alsa/pcm_plugin.h
-rw-r--r-- thomasanderson/primarygroup    11382 2019-11-05 15:50 ./usr/include/alsa/timer.h
-rw-r--r-- thomasanderson/primarygroup     6334 2019-11-05 15:50 ./usr/include/alsa/pcm_extplug.h
-rw-r--r-- thomasanderson/primarygroup     4220 2019-11-05 15:50 ./usr/include/alsa/mixer_abst.h
-rw-r--r-- thomasanderson/primarygroup    20071 2019-11-05 15:50 ./usr/include/alsa/pcm_old.h
-rw-r--r-- thomasanderson/primarygroup    14435 2019-11-05 15:50 ./usr/include/alsa/seqmid.h
-rw-r--r-- thomasanderson/primarygroup     8374 2019-11-05 15:50 ./usr/include/alsa/conf.h
drwxr-xr-x thomasanderson/primarygroup        0 2019-11-05 15:50 ./usr/include/alsa/sound/
-rw-r--r-- thomasanderson/primarygroup     5455 2019-11-05 15:50 ./usr/include/alsa/sound/hdspm.h
-rw-r--r-- thomasanderson/primarygroup    19790 2019-11-05 15:50 ./usr/include/alsa/sound/asoc.h
-rw-r--r-- thomasanderson/primarygroup     3040 2019-11-05 15:50 ./usr/include/alsa/sound/hdsp.h
-rw-r--r-- thomasanderson/primarygroup      368 2019-11-05 15:50 ./usr/include/alsa/sound/sscape_ioctl.h
-rw-r--r-- thomasanderson/primarygroup     4319 2019-11-05 15:50 ./usr/include/alsa/sound/asound_fm.h
-rw-r--r-- thomasanderson/primarygroup     3970 2019-11-05 15:50 ./usr/include/alsa/sound/sb16_csp.h
-rw-r--r-- thomasanderson/primarygroup    15320 2019-11-05 15:50 ./usr/include/alsa/sound/emu10k1.h
-rw-r--r-- thomasanderson/primarygroup     4537 2019-11-05 15:50 ./usr/include/alsa/sound/tlv.h
-rw-r--r-- thomasanderson/primarygroup      947 2019-11-05 15:50 ./usr/include/alsa/sound/type_compat.h
-rw-r--r-- thomasanderson/primarygroup     4748 2019-11-05 15:50 ./usr/include/alsa/pcm_rate.h
-rw-r--r-- thomasanderson/primarygroup     7154 2019-11-05 15:50 ./usr/include/alsa/rawmidi.h
-rw-r--r-- thomasanderson/primarygroup    34760 2019-11-05 15:50 ./usr/include/alsa/topology.h
-rw-r--r-- thomasanderson/primarygroup     5186 2019-11-05 15:50 ./usr/include/alsa/global.h
-rw-r--r-- thomasanderson/primarygroup    11795 2019-11-05 15:50 ./usr/include/alsa/seq_event.h
-rw-r--r-- thomasanderson/primarygroup    32477 2019-11-05 15:50 ./usr/include/alsa/seq.h
-rw-r--r-- thomasanderson/primarygroup    14402 2019-11-05 15:50 ./usr/include/alsa/mixer.h
-rw-r--r-- thomasanderson/primarygroup     3242 2019-11-05 15:50 ./usr/include/alsa/error.h
-rw-r--r-- thomasanderson/primarygroup    18191 2019-11-05 15:50 ./usr/include/alsa/asoundef.h
-rw-r--r-- thomasanderson/primarygroup     2390 2019-11-05 15:50 ./usr/include/alsa/seq_midi_event.h
-rw-r--r-- thomasanderson/primarygroup    28762 2019-11-05 15:50 ./usr/include/alsa/control.h
-rw-r--r-- thomasanderson/primarygroup     1899 2019-11-05 15:50 ./usr/include/alsa/pcm_external.h
-rw-r--r-- thomasanderson/primarygroup     7504 2019-11-05 15:50 ./usr/include/alsa/hwdep.h
drwxr-xr-x thomasanderson/primarygroup        0 2019-10-25 14:00 ./usr/include/pulse/
-rw-r--r-- thomasanderson/primarygroup     3065 2019-10-25 14:00 ./usr/include/pulse/subscribe.h
-rw-r--r-- thomasanderson/primarygroup     2071 2019-10-25 14:00 ./usr/include/pulse/glib-mainloop.h
-rw-r--r-- thomasanderson/primarygroup     3708 2019-10-25 14:00 ./usr/include/pulse/ext-device-restore.h
-rw-r--r-- thomasanderson/primarygroup    21959 2019-10-25 14:00 ./usr/include/pulse/volume.h
-rw-r--r-- thomasanderson/primarygroup     2371 2019-10-25 14:00 ./usr/include/pulse/version.h
-rw-r--r-- thomasanderson/primarygroup    41806 2019-10-25 14:00 ./usr/include/pulse/stream.h
-rw-r--r-- thomasanderson/primarygroup     1278 2019-10-25 14:00 ./usr/include/pulse/cdecl.h
-rw-r--r-- thomasanderson/primarygroup     6220 2019-10-25 14:00 ./usr/include/pulse/mainloop-api.h
-rw-r--r-- thomasanderson/primarygroup     2419 2019-10-25 14:00 ./usr/include/pulse/mainloop-signal.h
-rw-r--r-- thomasanderson/primarygroup     7605 2019-10-25 14:00 ./usr/include/pulse/pulseaudio.h
-rw-r--r-- thomasanderson/primarygroup     3610 2019-10-25 14:00 ./usr/include/pulse/gccmacro.h
-rw-r--r-- thomasanderson/primarygroup    13700 2019-10-25 14:00 ./usr/include/pulse/sample.h
-rw-r--r-- thomasanderson/primarygroup    13537 2019-10-25 14:00 ./usr/include/pulse/context.h
[... cut ...]
-rw-r--r-- thomasanderson/primarygroup     3052 2019-10-25 14:00 ./usr/include/pulse/timeval.h
-rw-r--r-- thomasanderson/primarygroup     2093 2019-10-25 14:00 ./usr/include/pulse/util.h
-rw-r--r-- thomasanderson/primarygroup     3915 2019-10-25 14:00 ./usr/include/pulse/ext-stream-restore.h
-rw-r--r-- thomasanderson/primarygroup    22235 2019-10-25 14:00 ./usr/include/pulse/proplist.h
```

It looks like the `-dev` header packages are already installed - so hopefully will require less work with `cmake` to get things building :)

mpromonet, 2020-12-17:
Hi gdhgdhgdh,

I update cmake to build with alsa/pulse available from sysroot provided by WebRTC SDK.
Did you try with latest release ?

Best Regards,
Michel.

closed 2021-01-01 by mpromonet