mpromonet/webrtc-streamer #371: Memory Leak

kasdovel, 2020-12-24:
**Describe the bug**
I think there is a memory leak in the program.
When I'm running the application on the first time (I'm using docker on linux server), the container takes about 4.3MiB of RAM.
When I'm connecting to the 'Bunny' camera [via the html page provided on port 8000], the containers takes about 11.6MiB of RAM when the usage is stable.
When I'm dicconnecting from the 'Bunny' camera, the RAM usage of the container goes down to about 9MiB, and never go below that.
I view how much RAM my container uses by running `docker stats {CONTAINER_ID}`.
The problem is when I'm connecting to alot of cameras [simultaneously or even watching one after another], the RAM usage is so high and my container crashes.

**To Reproduce**
Steps to reproduce the behavior:
1. Run this docker command: `docker run -d -p 8000:8000 -it mpromonet/webrtc-streamer`
2. Run `docker stats {CONTAINER_ID}` and look on the RAM usage
3. Go to `{SERVER_IP_ADDRESS}/8000`
4. Connect to the 'Bunny' camera
5. Look at the RAM usage
6. Disconnect from the 'Bunny' camera
7. Look at the RAM usage

**Expected behavior**
I would expect that when I disconnect from a camera, all of its related resources would be free.

**Screenshots**
[image]
Container stats on start

[image]
Container stats on 'Bunny' connection

[image]
Container stats on 'Bunny' disconnection

**Desktop (please complete the following information):**
My desktop is:
 - OS: Windows 10
 - Browser: Chrome
 - Version 87
My server is:
 - OS: Ubuntu
 - Version: 18.04.3

**Additional context**
Other than that, your application works great!

mpromonet, 2021-01-29:
Hi @kasdovel,

There is a memory leak reported by valgrind : 
```
==24620== 1,788 (440 direct, 1,348 indirect) bytes in 1 blocks are definitely lost in loss record 41 of 41
==24620==    at 0x483BE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==24620==    by 0x3930CC: PeerConnectionManager::CreatePeerConnection(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) (PeerConnectionManager.cpp:961)
==24620==    by 0x38E93E: PeerConnectionManager::call(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, Json::Value const&) (PeerConnectionManager.cpp:638)
```
I am trying to fix this.

Best Regards,
Michel.

mpromonet, 2021-02-21:
Hi @kasdovel 

I think https://github.com/mpromonet/webrtc-streamer/commit/6cc89577e31d99e91fdb5f93a0557c29eaa3d8d7 stop memory flooding. Then I close this issue.
If you still have memory leak, open a new one describing the problem and what you try to understand it (notice that virtual memory may not always reduce releasing memory)

Best Regards,
Michel.

closed 2021-02-21 by mpromonet

CHN-STUDENT, 2024-06-21:
@mpromonet hey mpromonet, today we found memory leak too, we use v0.6.5 on windows7, could you tell me https://github.com/mpromonet/webrtc-streamer/commit/6cc89577e31d99e91fdb5f93a0557c29eaa3d8d7 is which new version? I think we need to upgrade our os.

mpromonet, 2024-06-22:
Hi @CHN-STUDENT,
The commit https://github.com/mpromonet/webrtc-streamer/commit/6cc89577e31d99e91fdb5f93a0557c29eaa3d8d7 is integrated since v0.5.0, if this is the cause, it will not fix your problem.
Best Regards,
Michel.