mpromonet/webrtc-streamer #374: How to publish as a service and add a new video source？

apgk, 2021-01-04:
1. I want to add, delete and change the video source through API at any time

2. If you want to send the program as a window service

mpromonet, 2021-01-04:
Hi apgk,

In order to add, delete a video source, you just need to manage at client side. The server have a config to map names to url but you can use url directly and manage them in your javascript code. 

If you like to run as a windows service you can use simething like http://nssm.cc/ or others that create a service an run an executable.

Best Regards,
Michel.

apgk, 2021-01-04:
Examples：
[image]

code：`<html>
<head>
<script src="libs/adapter.min.js" ></script>
<script src="webrtcstreamer.js" ></script>
<script>        
    var webRtcServer      = null;
    window.onload         = function() { 
        webRtcServer      = new WebRtcStreamer("video",location.protocol+"//"+window.location.hostname+":8000");
	webRtcServer.connect("rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov");
    }
    window.onbeforeunload = function() { webRtcServer.disconnect(); }
</script>
</head>
<body> 
    <video id="video" />
</body>
</html>`

Is that so?Do you have any ready-made API examples？

mpromonet, 2021-01-04:
Hi,

Yes you can manage urls in whatever you like and use WebRtcStreamer.connect to connect a video tag to a webrtc stream.
The api is describe in http://htmlpreview.github.io/?https://github.com/mpromonet/webrtc-streamer-html/blob/master/jsdoc/WebRtcStreamer.html.
The webcomponent allow also this.

Best Regards,
Michel.

apgk, 2021-01-04:
Thanks Michel.
I know how to use it.

Thanks.

closed 2021-01-04 by apgk