mpromonet/webrtc-streamer #375: Docker restart because of audio device

gferandelle, 2021-01-07:
**Describe the bug**

While trying to launch the latest docker image I am facing to this error 

`# Failed to initialize the ADM.Version:v0.3.3-17-g5e5933f/Linux-x86_64 civetweb@v1.13 webrtc@9325d343e5-dirty live555helper@897450c
nullLogger level:3
[000:000][1] (audio_device_pulse_linux.cc:1553): failed to load symbol table
[000:000][1] (audio_device_pulse_linux.cc:144): failed to initialize PulseAudio
[000:000][1] (audio_device_impl.cc:342): Audio device initialization failed.
[000:000][1] (audio_device_pulse_linux.cc:1553): failed to load symbol table
[000:000][1] (audio_device_pulse_linux.cc:144): failed to initialize PulseAudio
[000:000][1] (audio_device_impl.cc:342): Audio device initialization failed.

#
# Fatal error in: ../../media/engine/adm_helpers.cc, line 39
# last system error: 0
# Check failed: 0 == adm->Init() (0 vs. -1)
`
So the docker is crashing and restarting.

The last working docker image I can use without problem is **sha256:882c6d935e52a3d1360e90aa095ebd4dafd8dd45a7a7ede0dcf99aecdf9b71ea**

**To Reproduce**

Digest of the image : **sha256:b05fbd12c06dd6ba3e6d3c446c9eb2d3e2cc42f92c71157ab23f8e1c3a0be97b**

Command used to launch the docker : 

/usr/bin/docker run -d --name="webrtc-streamer" \
                    --restart unless-stopped \
                    --cpus="2" \
		    --net=host \
                    -p 8443:8443 \
                    -v <ssl_hidden_path>:<ssl_hidden_path> \
                    -it mpromonet/webrtc-streamer:${TAG} \
                    -s<local_hidden_ip> \
                    -H8443s \
                    -c <hidden_pem_file>

**Env**
Docker on Ubuntu 18.04

**Context**

The stream (rtsp from axis cams) is called by a front app which is on an other machine.

Thanks for your help

mpromonet, 2021-01-07:
Hi,

Did you try to disable audio capture adding `-a` to the command line ?

Best Regards,
Michel.

gferandelle, 2021-01-07:
Hi,

I will try asap. But, looking at the Dockerfile, isn't it set by default in CMD ?

Regards.

gferandelle, 2021-01-08:
So I have tried to manually add `-a`. 

The container is stopping, but it looks like there is no error  (launched with -vvv):
```

Version:v0.3.3-17-g5e5933f/Linux-x86_64 civetweb@v1.13 webrtc@9325d343e5-dirty live555helper@897450c
nullLogger level:0
[000:000][1] (audio_device_impl.cc:76): Create
[000:000][1] (audio_device_impl.cc:84): CreateForTest
[000:000][1] (audio_device_buffer.cc:65): AudioDeviceBuffer::ctor
[000:000][1] (audio_device_impl.cc:122): AudioDeviceModuleImpl
[000:000][1] (audio_device_impl.cc:126): CheckPlatform
[000:000][1] (audio_device_impl.cc:137): current platform is Linux
[000:000][1] (audio_device_impl.cc:156): CreatePlatformSpecificObjects
[000:000][1] (audio_device_impl.cc:948): PlatformAudioLayer
[000:000][1] (audio_device_impl.cc:267): PulseAudio support is enabled.
[000:000][1] (audio_device_impl.cc:300): Dummy Audio APIs will be utilized.
[000:000][1] (audio_device_impl.cc:313): AttachAudioBuffer
[000:000][1] (audio_processing_impl.cc:274): Injected APM submodules:
Echo control factory: 0
Echo detector: 0
Capture analyzer: 0
Capture post processor: 0
Render pre processor: 0
[000:000][1] (webrtc_voice_engine.cc:263): WebRtcVoiceEngine::WebRtcVoiceEngine
[000:000][1] (webrtc_video_engine.cc:635): WebRtcVideoEngine::WebRtcVideoEngine()
[000:001][1] (webrtc_voice_engine.cc:285): WebRtcVoiceEngine::Init
[000:001][1] (webrtc_voice_engine.cc:293): Supported send codecs in order of preference:
[000:001][1] (webrtc_voice_engine.cc:296): opus/48000/2 { minptime=10 useinbandfec=1 } (111)
[000:001][1] (webrtc_voice_engine.cc:296): ISAC/16000/1 (103)
[000:001][1] (webrtc_voice_engine.cc:296): ISAC/32000/1 (104)
[000:001][1] (webrtc_voice_engine.cc:296): G722/8000/1 (9)
[000:001][1] (webrtc_voice_engine.cc:296): ILBC/8000/1 (102)
[000:001][1] (webrtc_voice_engine.cc:296): PCMU/8000/1 (0)
[000:001][1] (webrtc_voice_engine.cc:296): PCMA/8000/1 (8)
[000:001][1] (webrtc_voice_engine.cc:296): CN/32000/1 (106)
[000:001][1] (webrtc_voice_engine.cc:296): CN/16000/1 (105)
[000:001][1] (webrtc_voice_engine.cc:296): CN/8000/1 (13)
[000:001][1] (webrtc_voice_engine.cc:296): telephone-event/48000/1 (110)
[000:001][1] (webrtc_voice_engine.cc:296): telephone-event/32000/1 (112)
[000:001][1] (webrtc_voice_engine.cc:296): telephone-event/16000/1 (113)
[000:001][1] (webrtc_voice_engine.cc:296): telephone-event/8000/1 (126)
[000:001][1] (webrtc_voice_engine.cc:299): Supported recv codecs in order of preference:
[000:001][1] (webrtc_voice_engine.cc:302): opus/48000/2 { minptime=10 useinbandfec=1 } (111)
[000:001][1] (webrtc_voice_engine.cc:302): ISAC/16000/1 (103)
[000:001][1] (webrtc_voice_engine.cc:302): ISAC/32000/1 (104)
[000:001][1] (webrtc_voice_engine.cc:302): G722/8000/1 (9)
[000:001][1] (webrtc_voice_engine.cc:302): ILBC/8000/1 (102)
[000:001][1] (webrtc_voice_engine.cc:302): PCMU/8000/1 (0)
[000:001][1] (webrtc_voice_engine.cc:302): PCMA/8000/1 (8)
[000:001][1] (webrtc_voice_engine.cc:302): CN/32000/1 (106)
[000:001][1] (webrtc_voice_engine.cc:302): CN/16000/1 (105)
[000:001][1] (webrtc_voice_engine.cc:302): CN/8000/1 (13)
[000:001][1] (webrtc_voice_engine.cc:302): telephone-event/48000/1 (110)
[000:001][1] (webrtc_voice_engine.cc:302): telephone-event/32000/1 (112)
[000:001][1] (webrtc_voice_engine.cc:302): telephone-event/16000/1 (113)
[000:001][1] (webrtc_voice_engine.cc:302): telephone-event/8000/1 (126)
[000:001][1] (audio_device_impl.cc:333): Init
[000:001][1] (audio_device_impl.cc:677): SetPlayoutDevice(0)
[000:001][1] (adm_helpers.cc:44): Unable to set playout device.
[000:001][1] (audio_device_impl.cc:852): RegisterAudioCallback
[000:001][1] (audio_device_buffer.cc:82): RegisterAudioCallback
[000:001][1] (webrtc_voice_engine.cc:381): WebRtcVoiceEngine::ApplyOptions: AudioOptions {aec: 1, agc: 1, ns: 1, hf: 1, swap: 0, audio_jitter_buffer_max_packets: 200, audio_jitter_buffer_fast_accelerate: 0, audio_jitter_buffer_min_delay_ms: 0, audio_jitter_buffer_enable_rtx_handling: 0, typing: 1, experimental_agc: 0, experimental_ns: 0, residual_echo_detector: 1, }
[000:001][1] (audio_device_impl.cc:868): BuiltInAECIsAvailable
[000:001][1] (audio_device_generic.cc:18): BuiltInAECIsAvailable: Not supported on this platform
[000:001][1] (audio_device_impl.cc:871): output: 0
[000:001][1] (audio_device_impl.cc:884): BuiltInAGCIsAvailable
[000:001][1] (audio_device_generic.cc:28): BuiltInAGCIsAvailable: Not supported on this platform
[000:001][1] (audio_device_impl.cc:887): output: 0
[000:001][1] (audio_device_impl.cc:900): BuiltInNSIsAvailable
[... 87 lines cut ...]
[000:001][1] (audio_device_impl.cc:883): BuiltInAGCIsAvailable
[000:001][1] (audio_device_generic.cc:28): BuiltInAGCIsAvailable: Not supported on this platform
[000:001][1] (audio_device_impl.cc:886): output: 0
[000:001][1] (audio_device_impl.cc:899): BuiltInNSIsAvailable
[000:001][1] (audio_device_generic.cc:38): BuiltInNSIsAvailable: Not supported on this platform
[000:001][1] (audio_device_impl.cc:902): output: 0
[000:001][1] (webrtc_voice_engine.cc:491): Stereo swapping enabled? 0
[000:001][1] (webrtc_voice_engine.cc:496): NetEq capacity is 200
[000:001][1] (webrtc_voice_engine.cc:502): NetEq fast mode? 0
[000:001][1] (webrtc_voice_engine.cc:508): NetEq minimum delay is 0
[000:001][1] (webrtc_voice_engine.cc:514): NetEq handle reordered packets? 0
[000:001][1] (webrtc_voice_engine.cc:534): Experimental ns is enabled? 0
[000:001][1] (webrtc_voice_engine.cc:585): NS set to 1
[... cut ...]
[000:001][1] (matched_filter.cc:455): Filter 0: start: 0 ms, end: 128 ms.
[000:001][1] (matched_filter.cc:455): Filter 1: start: 96 ms, end: 224 ms.
[000:001][1] (matched_filter.cc:455): Filter 2: start: 192 ms, end: 320 ms.
[000:001][1] (matched_filter.cc:455): Filter 3: start: 288 ms, end: 416 ms.
[000:001][1] (matched_filter.cc:455): Filter 4: start: 384 ms, end: 512 ms.
[000:001][1] (agc_manager_direct.cc:69): [agc] GetMinMicLevel
[000:001][1] (agc_manager_direct.cc:73): [agc] Using default min mic level: 12
HTTP Listen at 8443s
```

mpromonet, 2021-01-09:
Hi,

The -a option seems to solve the issue, isn't it ?

Best Regards,
Michel.

gferandelle, 2021-01-09:
No, it is not. The docker stop to run without any error logs  after it says listening on the HTTP port

mpromonet, 2021-01-09:
Hi,

I reproduce the problem, but this is not link to audio cpature, the http server crash starting with ssl.

Best Regards,
Michel.

mpromonet, 2021-01-09:
Hi,

With version v0.3.3-24 initialization of ssl seems fixed. I also manage the audio capture initialization to avoid crash when initialization fails.

Best Regards,
Michel

closed 2021-01-13 by mpromonet