mpromonet/webrtc-streamer #376: Libcrypto error with Docker

gferandelle, 2021-01-20:
**Describe the bug**

Hi @mpromonet,
Best wishes for this new year.

I am back from my hollydays, so I just tried your last docker image. I am facing to an issue you already worked on [https://github.com/mpromonet/webrtc-streamer/issues/248](issue #248)

```
[000:002][1] (render_delay_buffer.cc:378): Applying total delay of 5 blocks.
[000:002][1] (matched_filter.cc:455): Filter 0: start: 0 ms, end: 128 ms.
[000:002][1] (matched_filter.cc:455): Filter 1: start: 96 ms, end: 224 ms.
[000:002][1] (matched_filter.cc:455): Filter 2: start: 192 ms, end: 320 ms.
[000:002][1] (matched_filter.cc:455): Filter 3: start: 288 ms, end: 416 ms.
[000:002][1] (matched_filter.cc:455): Filter 4: start: 384 ms, end: 512 ms.
[000:002][1] (agc_manager_direct.cc:69): [agc] GetMinMicLevel
[000:002][1] (agc_manager_direct.cc:73): [agc] Using default min mic level: 12
HTTP Listen at 8443s
initialize_ssl: error loading library libcrypto.so
Error initializing SSL context
Cannot Initialize start HTTP server exception:null context when constructing CivetServer. Possible problem binding to port.
Exit
```

**Desktop (please complete the following information):**
Ubuntu 18.04 with docker 19.03.13
Docker image : sha256:0d11bd8916b7d8f40706cdee2bfa9e6a16942b52cd59ce11a483da5be9b28e40

AlpenFlizzer, 2021-01-20:
Hi!

I am also facing the same issue with the latest pull of your image.

Regards, Sascha

mpromonet, 2021-01-20:
Hi,

Using latest docker should now fix the problem of ssl dependency.
`Version:v0.3.3-29-g2aa8aa2/Linux-x86_64` is running.

Best Regards,
Michel.

AlpenFlizzer, 2021-01-21:
I can confirm it is working now using the latest tag on an amd64 machine.

BUT: While v0.3.3 worked with RSA1024 private key length, the new version wont work with such a small key. I tested with an RSA4096 key (self signed) and it works. (Note, that the documentation of civetweb gives example code to create the necessary .pem using a RSA1024 key...)

Best, Sascha

mpromonet, 2021-01-21:
Hi Sascha,

Thanks for the feedback.
I also upgrade the base image from ubuntu18.04 to ubuntu20.04 this upgrade openssl and then maybe crypto constrains.

Best Regards,
Michel.

closed 2021-01-21 by mpromonet

AlpenFlizzer, 2021-01-23:
Hi!

I have to reopen the issue.

The problem seems to exist on ARM for the latest build and also for v0.3.3

Anyone else having the same problem on Raspberry Pi 3b?

Best, Sascha

mpromonet, 2021-01-23:
Hi @AlpenFlizzer 

It should be fixed since this commit https://github.com/mpromonet/webrtc-streamer/commit/2ffaa3ba1eb3ff8df65899a8233084c8e38854cf
I checked on Rasberry Pi2 that use the arm/v7 image, but I don't know which arm variant use the Raspberry Pi3 ? maybe the arm64 ?

Best Regards,
Michel.

AlpenFlizzer, 2021-01-23:
Hi!

I checked your latest builds on Raspi 2 and Raspi 3, both working fine now! Thank you very much for your effort!

Best, Sascha