mpromonet/webrtc-streamer #378: while connecting to Jitsi, browser console show me "Strophe is disconnected."

higunjan, 2021-01-27:
**Describe the bug**
Facing the issue of "Strophe is disconnected." while connect to Jitsi.
[image]

**To Reproduce**
Simple follow the code which you've provided and try to execute in browser.

**Expected behavior**
It's should work as your working demo with any one dummy video.

**Code**
```
<!-- Mentioned all libs loaded on folders -->
<html>
<head>
<script src="libs/strophe.min.js" ></script>
<script src="libs/strophe.muc.min.js" ></script>
<script src="libs/strophe.disco.min.js" ></script>
<script src="libs/strophe.jingle.sdp.js"></script>
<script src="libs/jquery-2.1.1.min.js"></script>
<script src="xmppvideoroom.js" ></script>
<script>        
	var xmpp = new XMPPVideoRoom("meet.jit.si", null);
	xmpp.join("rtsp", "rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov","testroom");
</script>
</head>
</html>
```

**Where deployed:**
Using Node 12.14.x LTS and express server used to server the above html page on xyz server

**Additional context**
I'm not sure why demo is not working as mentioned in you README file.

Please can you elaborate what else I'm missing here.

mpromonet, 2021-01-27:
Hi,

It may depends on the version you are using. 

I tried the html you gave served running `python3 -m  http.server`, it connect to Strophe. It doesnot make more because the null parameter indicate to use same http server for webrtc api.
Something that is working a little more could be 
```
<html>
<head>
<script src="libs/strophe.min.js" ></script>
<script src="libs/strophe.muc.min.js" ></script>
<script src="libs/strophe.disco.min.js" ></script>
<script src="libs/strophe.jingle.sdp.js"></script>
<script src="libs/jquery-2.1.1.min.js"></script>
<script src="xmppvideoroom.js" ></script>
<script>
        var xmpp = new XMPPVideoRoom("meet.jit.si", <url of webrtc-streamer> );
        xmpp.join("testroom", "Bunny" , "jitiuser"+Math.random());
</script>
</head>
</html>
```
But clearly publishing to jitsi need improvement & debug...

Best Regards,
Michel.

higunjan, 2021-01-28:
Michel, Thank you.

API url I understood can we deploy that API server our own side. where is the code ? `<url of webrtc-streamer>`

Instead of `rtsp://` URL you're using the NAME like `Bunny`, what's the reason behind that, why can't we provide the `rtsp://` stream?

closed 2021-02-28 by mpromonet