mpromonet/webrtc-streamer #379: RTSP parameters lost

gferandelle, 2021-02-03:
**Describe the bug**
I don't have logs about it, but I have a different behavior using the latest docker images compared an old one (0.2.7).

We are using Axis cameras with this kind of request :

`rtsp://192.168.1.103:554/axis-media/media.amp?videocodec=h264&resolution=480x270&compression=30&fps=20`

With image v0.2.7, parameters like resolution, compression, etc are fine used.

But with latest version, it is like parameters are lost.

This is what I can extract from logs that looks like relevant 

`Sending request: PLAY rtsp://192.168.1.104:554/axis-media/media.amp?videocodec=h264 RTSP/1.0`

mpromonet, 2021-02-03:
Hi,

This may depends how you encode the url ?

Best Regards,
Michel.

gferandelle, 2021-02-04:
We never changed the way to encode it and it works with image v0.2.7. 

This is an example of encoded url

`rtsp%3A%2F%2F192.168.1.110%3A554%2Faxis-media%2Fmedia.amp%3Fcamera%3D2%26videocodec%3Dh264%26resolution%3D480x270%26compression%3D30%26fps%3D20`

mpromonet, 2021-02-04:
Hi,

Then you may look to difference between v0.2.7, feel free to submit your proposal.

Best Regards,
Michel.

commit 5765b350 (2021-02-04): fix #379: disable decode_url of civetweb
closed 2021-02-04 by mpromonet