mpromonet/webrtc-streamer #380: It needs hardware h264 support

imerso, 2021-02-06:
Nice project you have here, but the software re-encoding makes it too slow for higher resolution streaming. Even simple hd 1280x720 almost kills the poor rpi2 already.

It definitely needs to support hardware h264 directly!

It is "just" a matter of bypassing that unnecessary re-encoding. There is a project which patches the WebRTC code to make it accept raw h264 (which is what the rpi camera provides, for example), eliminating the very slow software re-encoding:

https://github.com/nicotyze/Webrtc-H264Capturer

I am willing to help this project to get hw h264 integrated. Such an improvement would make it so much better! Imagine hd or even full-hd h264 at steady 30fps using less than 50% of a Raspberry CPU. That is perfectly possible -- I do have my own native h264 streamer which can do that (it is not WebRTC compatible, though) for years now. Mine streams raw h264 to an also native Android app and it is blazingly fast, no re-encodings at all. I am pretty much sure this can be done to make WebRTC blazingly fast as well.

Although I am an experienced programer, I don't know the WebRTC codebase yet, so I can't say how much time I would need to make such an integration.

Here is the idea, though. I will be looking into WebRTC code to see how hard it really would be to make the needed changes, and I am also looking for comments and accepting/offering help.

Thanks.

mpromonet, 2021-02-06:
Hi imerso,

This is a reccurent will to reduce CPU usage.
If the purpose is for raspberry target, a possible way is like https://github.com/kclyu/rpi-webrtc-streamer does, adding H264 encoder/decoder to the WebRTC codec factory. 

Best Regards,
Michel.

mpromonet, 2021-03-07:
Hi imerso,

Implementing NullDecoder & NullEncoder seems a possible way, it's really early testing.

You may try to run webrtc-streamer with `-o` option, this will store RTP H264 payload frame encoded and forward it to WebRTC. It use less ressource but   is less adaptative (no resize, no bandwidth adaptation, no codec adaptation).

Best Regards,
Michel.

imerso, 2021-03-16:
This is awesome, thank you for your efforts!

I was delaying this reply because I wanted to try it before commenting, but I moved to a new home and I'm also so overwhelmed by daily work that I simply did not manage to mount a pi and try it out.

Will do that soon, thank you!

prncvrm, 2021-03-31:
@imerso @mpromonet i was trying my built binary for macos https://github.com/mpromonet/webrtc-streamer/issues/391 and saw high cpu usage ~200% on [htop](https://htop.dev/), (the application named web-streamer),
[image]
tried using null codec, with -o flag, ran into the following error,
```
[020:610][10243] (video_stream_encoder.cc:1784): Failed to encode frame. Error code: -4
[020:662][10243] (video_stream_encoder.cc:1784): Failed to encode frame. Error code: -4
[020:716][34575] (video_stream_encoder.cc:1784): Failed to encode frame. Error code: -4
[020:770][34575] (video_stream_encoder.cc:1784): Failed to encode frame. Error code: -4
[020:823][34575] (video_stream_encoder.cc:1784): Failed to encode frame. Error code: -4
[020:878][34575] (video_stream_encoder.cc:1784): Failed to encode frame. Error code: -4
[020:935][10243] (video_stream_encoder.cc:1784): Failed to encode frame. Error code: -4
[020:994][34575] (video_stream_encoder.cc:1784): Failed to encode frame. Error code: -4
[021:047][34575] (video_stream_encoder.cc:1784): Failed to encode frame. Error code: -4
[021:102][34575] (video_stream_encoder.cc:1784): Failed to encode frame. Error code: -4
```
i debugged more and found error code -4 is `WEBRTC_VIDEO_CODEC_ERR_PARAMETER`. Could you guide me here?. Works pretty fine without -o flag though. its just eating too much cpu.

mpromonet, 2021-03-31:
Hi @prncvrm

-o option use NullDecoder & NullEncoder storing encoded frame with `webrtc::VideoFrameBuffer::Type::kNative` type. I guess if NullEncoder::Encode return WEBRTC_VIDEO_CODEC_ERR_PARAMETER, it means frametype is not `webrtc::VideoFrameBuffer::Type::kNative`

Best Regards,
Michel.

mpromonet, 2021-04-05:
Hi imerso,

I am closing this issue to clean-up, I made a simple implementation of an V4L2 H264 capture that post encoded frame with -o option, like RTSP does.
 
Best Regards,
Michel.

closed 2021-04-05 by mpromonet

imerso, 2021-07-08:
@mpromonet I finally managed to do a proper experiment, and it worked like a charm. Congratulations, you did it! Thanks!