mpromonet/webrtc-streamer #384: Session restart after ~60secs

gferandelle, 2021-02-19:
**Describe the bug**
Hi Michel, we are facing to a RTSP sessions restart periodically (around every 60 secondes). In logs I can see : 

```
[110:961][69] (rtcp_receiver.cc:310): OnPeriodicRttUpdate: Timeout: No increase in RTCP RR extended highest sequence number.
[111:816][61] (rtcp_receiver.cc:310): OnPeriodicRttUpdate: Timeout: No increase in RTCP RR extended highest sequence number.
[112:022][76] (rtcp_receiver.cc:310): OnPeriodicRttUpdate: Timeout: No increase in RTCP RR extended highest sequence number.
[116:815][61] (rtcp_receiver.cc:308): OnPeriodicRttUpdate: Timeout: No RTCP RR received.
```
After that, a new session is started, but without audio.

**Screenshots**
[image]

**Additional context**
running under docker latest image

mpromonet, 2021-02-24:
Hi,

Maybe increasing the log level could help you to understand the root cause that close the connection, the webrtc stream ? the RTSP/RTP stream ?

Best Regards;
Michel.

gferandelle, 2021-02-24:
I will try to increase logs tomorrow. But accordingly some research on internet about those errors, it looks like they are caused by missing keepalive kind messages that are not sent by the RTSP client. But I am not sure.

gferandelle, 2021-02-25:
I don't really know what to search on logs. I don't know if it is related to it, but sometimes there is no value for the parameter **pings_since_last_response** on logs linked to connection.cc.

```
[118:470][8] (connection.cc:791): Conn[600aae50:0:Net[enp0s31f6:172.16.8.x/28:Unknown:id=1]:nklmr6WC:1:0:local:udp:172.16.8.x:55683->hw4ycTRw:1:1677729535:stun:udp:10.254.0.x:56734|C
RWS|S|1|0|7205793488602807806|32]: UpdateState(), ms since last received response=1469, ms since last received data=877, rtt=100, pings_since_last_response=
```

mpromonet, 2021-02-28:
some firewall rules filtering udp ?

gferandelle, 2021-03-01:
I will check again. We have an UFW firewall which is running on the docker host. Whereas everything is open, cause we use it only for port forwarding. I will check if I missed something. 

To be sure, RR packets are sent between the streamer and the camera, right ?

mpromonet, 2021-03-01:
Hi @gferandelle 

Yes rtcp_from_spec.c is live555 code, your log seems to indicate that RTCP Receiver Report are not received by webrtc-streamer. Using RTP over TCP may helps you ?

Best Regards,
Michel.

closed 2021-07-04 by mpromonet

jianxin123, 2022-12-14:
Hello, I seem to have the same problem. How did you solve it?