mpromonet/webrtc-streamer #387: How can I change the resolution of RTSP

QiuanLee, 2021-03-06:
We are developing a program to push RTSP stream to video room in the weak network environment, and everything is going well at present. However, we find that the resolution of RTSP video is too high. How can we modify the resolution of RSTP (without modifying the configuration of the original camera)

mpromonet, 2021-03-06:
Hi,

You can change the WebRTC stream using the options width & height.
For instance https://webrtc-streamer.herokuapp.com/webrtcstreamer.html?video=Western%20Cape&options=rtptransport%3Dtcp%26timeout%3D60%26width%3D320
These options could be set in different way, in the webcomponent or in the javascript or in the webrtcstreamer.json

Best Regards,
Michel.

Syn4ptic, 2021-03-11:
HI Michel,

Any idea why my stream is not reacting on change resolution ? 

Just before launching app by typing **4l2-ctl -V** i can see below  config:

Format Video Capture:
        Width/Height      : 1280/720
        Pixel Format      : 'H264' (H.264)
        Field             : None
        Bytes per Line    : 0
        Size Image        : 921600
        Colorspace        : SMPTE 170M

In browser i can see "mmal service 16.1" stream but in tiny window. Adding any options is not affecting stream at all. When app is stooped **4l2-ctl -V** shows me: 

Format Video Capture:
        Width/Height      : 128/96
        Pixel Format      : 'YU12' (Planar YUV 4:2:0)
        Field             : None
        Bytes per Line    : 128
        Size Image        : 18432
        Colorspace        : SMPTE 170M

Running webrtc-streamer-v0.3.5-Linux-armv6l-Release on Pi Zero. Any tips will be appreciated and  BTW great project :)
Regards 
Marcin

mpromonet, 2021-03-11:
Hi Marcin,

The capture device choose the resolution and then it keep this resolution.   
In order to set the resolution, you should close all streams (you can check opened stream using /admin.html page), and set options to ask capture to select resolution (by the way, prefered capture format seems i420). 

Best Regards,
Michel.

QiuanLee, 2021-03-12:
HI Michel,

Thank you for your reply. I studied the example you provided, but this example does not conform to the specific situation of our production environment. Our peer-to-peer environment is a complex private network environment with large network delay and speed limit. It is only suitable for Janus related services. Thank you for your help!

Best Regards,
QiuanLee

mpromonet, 2021-03-13:
Hi QiuanLee,

Then I close the issue that was already use to speak about a different case related to V4L2 capture. I did not really understand your situation, you can reduce the resolution of the output stream, the input depend on your camera capabilities.

Best Regards,
Michel.

closed 2021-03-13 by mpromonet