mpromonet/webrtc-streamer #388: Stream fails to start on rpi arm7/debian x86, but succeeds on macOS

mafrosis, 2021-03-06:
**Describe the bug**

I can connect and stream RTSP perfectly on my macbook, but trying the same on my target rpi4 fails. Similarly a test on debian x86 results in the same failure.

The tests are all LAN only, so as I understand it this cannot be a STUN/TURN issue. The camera is a Foscam X1, with no direct internet access.

I've read that others have this working on linux/rpi, so it's either a bug, or a configuration issue on my part.

**To Reproduce**

```
docker run -p 8000:8000 -it -v /tmp/webrtc-streamer.json:/app/config.json mpromonet/webrtc-streamer -H 0.0.0.0:8000 -C config.json -vv -s-
```

*webrtc-streamer.json*:
```
{
	"urls": {
		"home": {
			"video": "rtsp://hass:supersecret@192.168.20.222:88/videoMain"
		}
	}
}
```

**Expected behavior**

I expect to be able to access the stream at `localhost:8000` on my macbook, or `<hostname>:8000` when running this docker image on another server on the LAN.

**Desktop (please complete the following information):**

Chrome 88 (same results seen in Firefox 86)

**Additional context**

<details>
  <summary>Logs from macOS, where streaming is *successful*:</summary>
  
```
Version:v0.3.5-15-g50a462f/Linux-x86_64 civetweb@v1.13 webrtc@1e2da374b3-dirty live555helper@897450c
{
	"urls" :
	{
		"home" :
		{
			"video" : "rtsp://hass:eggsbacon-1@192.168.20.222:88/videoMain"
		}
	}
}Logger level:1
[000:000][1] (audio_device_buffer.cc:64): AudioDeviceBuffer::ctor
[000:001][1] (audio_device_impl.cc:137): current platform is Linux
[000:001][1] (audio_device_impl.cc:156): CreatePlatformSpecificObjects
[000:001][1] (audio_device_impl.cc:948): PlatformAudioLayer
[000:001][1] (audio_device_impl.cc:267): PulseAudio support is enabled.
[000:001][1] (audio_device_impl.cc:272): Linux PulseAudio APIs will be utilized
[000:001][1] (audio_device_impl.cc:313): AttachAudioBuffer
[000:001][1] (audio_device_buffer.cc:180): SetRecordingSampleRate(0)
[000:001][1] (audio_device_buffer.cc:186): SetPlayoutSampleRate(0)
[000:001][1] (audio_device_buffer.cc:200): SetRecordingChannels(0)
[000:001][1] (audio_device_buffer.cc:206): SetPlayoutChannels(0)
[000:001][1] (audio_device_impl.cc:333): Init
[000:006][1] (audio_device_pulse_linux.cc:1610): failed to connect context, error=-1
[000:006][1] (audio_device_pulse_linux.cc:144): failed to initialize PulseAudio
[000:006][1] (audio_device_impl.cc:342): Audio device initialization failed.
[000:006][1] (PeerConnectionManager.cpp:180): audio init fails -> disable audio capture
[000:006][1] (audio_processing_impl.cc:274): Injected APM submodules:
Echo control factory: 0
Echo detector: 0
Capture analyzer: 0
Capture post processor: 0
[... 1887 lines cut ...]
[007:668][73] (resource_adaptation_processor.cc:132): Registered resource "EncoderUsageResource".
[007:668][73] (quality_scaling_experiment.cc:39): QP thresholds: low: 29, high: 95
[007:668][73] (quality_scaler.cc:222): QP thresholds: low: 29, high: 95
[007:668][73] (resource_adaptation_processor.cc:132): Registered resource "QualityScalerResource".
[007:668][73] (video_stream_encoder.cc:1524): Dropping frame. Too large for target bitrate.
[007:668][73] (video_stream_adapter.cc:488): Scaling down resolution, max pixels: 552960
[007:668][73] (video_stream_encoder_resource_manager.cc:652): Downgrade counts: fps: {quality:0cpu:0}, resolution {quality:1cpu:0}
[007:668][73] (video_stream_encoder.cc:2124): Updating sink restrictions from QualityScalerResource to { max_pixels_per_frame=552960 }
[007:668][73] (video_stream_encoder.cc:1524): Dropping frame. Too large for target bitrate.
[007:668][1] (video_source_sink_controller.cc:83): Pushing SourceSink restrictions: max_fps=60 max_pixel_count=552960 target_pixel_count=null resolutions={1280x720}
[007:735][73] (video_stream_encoder.cc:1524): Dropping frame. Too large for target bitrate.
[007:791][73] (video_stream_encoder.cc:1524): Dropping frame. Too large for target bitrate.
[008:127][10] (network.cc:912): Connect failed with 99
[010:128][10] (network.cc:912): Connect failed with 99
[012:128][10] (network.cc:912): Connect failed with 99
[014:131][10] (network.cc:912): Connect failed with 99
[016:133][10] (network.cc:912): Connect failed with 99
[018:135][10] (network.cc:912): Connect failed with 99
[020:138][10] (network.cc:912): Connect failed with 99
^CSIGINT
[021:164][1] (audio_device_impl.cc:319): ~AudioDeviceModuleImpl
[021:164][1] (audio_device_buffer.cc:75): AudioDeviceBuffer::~dtor
Exit
```
</details>

mpromonet, 2021-03-06:
Hi,

It seems the ICE resolution is failing : 
`
[006:215][10] (p2p_transport_channel.cc:1260): Failed to resolve ICE candidate hostname 36da9538-7734-4ba5-a418-a2f954ece6c8.local with error -2`

Did you try to run docker with `-v /var/run/dbus:/var/run/dbus` ? or `-net=host` ?

Best Regards,
Michel.

mafrosis, 2021-03-07:
Hey. Thanks for taking a look.

Same result using the `dbus` suggestion:
```
docker run -p 8000:8000 -it -v /tmp/webrtc-streamer.json:/app/config.json -v /var/run/dbus:/var/run/dbus mpromonet/webrtc-streamer -H 0.0.0.0:8000 -C config.json -vv -s-
```

However, running with `--net=host` works fine.. Why is that? I have to admit, in general I much prefer port mapping over running every container on the host network, but exceptions can be made to make important videos streamable 😄

mpromonet, 2021-03-07:
Hi mafrosis,

I think in your case it should work using a STUN, if you did not disable it, running:
`docker run -p 8000:8000 -v /tmp/webrtc-streamer.json:/app/config.json mpromonet/webrtc-streamer`

If you have no internet connection you may look to some solution like running an STUN (like coturn in your network) and run 
`docker run -p 8000:8000 -v /tmp/webrtc-streamer.json:/app/config.json mpromonet/webrtc-streamer -s<stun address>`

Best Regards,
Michel.

mafrosis, 2021-04-06:
Hi @mpromonet I'm happy to close this as I now just run this container with `--net=host`. Thanks for the software 💯