mpromonet/webrtc-streamer #389: I can not hear audio when play rtsp video source

jw1216663169, 2021-03-16:
**Describe the bug**
I can not hear audio when play rtsp video source，but use VLC media playerit can work

**To Reproduce**
Steps to reproduce the behavior:

//let webrtcUrl = location.protocol+"//"+window.location.hostname+":8000"
let webrtcUrl = "http://192.168.8.250:8000"
let rtspURL ="rtsp://admin:admin12345@192.168.8.12:554/cam/realmonitorchannel=1%26subtype=0"
let id = "video" + num;

if(!this.videos[id]){
    this.videos[id] = new WebRtcStreamer(id ,webrtcUrl);
}
this.videos[id].connect(rtspURL);

**Screenshots**
[74082:475][2585] (PeerConnectionManager.h:229): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.3544739833956314
[74082:479][2585] (PeerConnectionManager.h:235): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:3 peerid:0.3544739833956314
Created new TCP socket 11 for connection
Created new TCP socket 12 for connection
[74082:481][2585] (PeerConnectionManager.h:229): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.3544739833956314
[74082:483][2585] (PeerConnectionManager.h:235): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:0 peerid:0.3544739833956314
Start playing sink for "video/H264" subsession
markerSize:4
[74082:730][10903] (liveaudiosource.h:98): LiveAudioSource::onNewSession not support codecm=audio 0 RTP/AVP 97
a=control:trackID=1
a=rtpmap:97 MPEG4-GENERIC/8000
a=fmtp:97 streamtype=5;profile-level-id=1;mode=AAC-hbr;sizelength=13;indexlength=3;indexdeltalength=3;config=1588
a=recvonly

[74082:858][2585] (PeerConnectionManager.h:144): virtual void PeerConnectionManager::DataChannelObserver::OnStateChange() channel:ServerDataChannel state:open
[74082:859][2585] (PeerConnectionManager.h:225): virtual void PeerConnectionManager::PeerConnectionObserver::OnDataChannel(rtc::scoped_refptr<webrtc::DataChannelInterface>)
[74082:859][2585] (PeerConnectionManager.h:144): virtual void PeerConnectionManager::DataChannelObserver::OnStateChange() channel:ClientDataChannel state:open
[74082:860][2585] (PeerConnectionManager.h:151): virtual void PeerConnectionManager::DataChannelObserver::OnMessage(const webrtc::DataBuffer &) channel:ClientDataChannel msg:local channel openned
[74082:860][2585] (PeerConnectionManager.h:151): virtual void PeerConnectionManager::DataChannelObserver::OnMessage(const webrtc::DataBuffer &) channel:ServerDataChannel msg:remote channel openned

mpromonet, 2021-03-16:
Hi @jw1216663169 

As we can see from your log 

> [74082:730][10903] (liveaudiosource.h:98): LiveAudioSource::onNewSession not support codec

MPEG4-GENERIC/8000 is not implemented only PCM & opus are supported.

Best Regards,
Michel.

jw1216663169, 2021-03-19:
OK，thank u!

closed 2021-03-19 by mpromonet

win2023PHe, 2025-09-15:
@mpromonet hi sir, i need to play aac video ,how can i do