mpromonet/webrtc-streamer #390: How to specify Range Clock on RTSP stream

s-beard, 2021-03-17:
Hi Michel,

I want to playback a video recorded by my NVR by specify a Range: Clock in absolute time. I use your webrtc-streamer to play the live from RTSP url like rtsp://192.168.1.124:554/camera2
But i search, how can i use your webrtc-streamer to play my video from an absolute date time. I found an older post on live555 about this. 
https://live-devel.live555.narkive.com/esUSI3qm/how-to-specify-range-in-rtspclient-that-are-not-using-npt

Do you find a solution to push the range clock information to lthe RTSP server ?

Thanks,

s-beard, 2021-03-17:
Hi,
I have tested to replace the 
	this->sendPlayCommand(*m_session, continueAfterPLAY);
By the new one,
	const char* absStartTime = "20210316T130000Z";
	const char* absEndTime = "20210316T140000Z";
	this->sendPlayCommand(*m_session, continueAfterPLAY, absStartTime, absEndTime);
I don't have any errors but the vidéo played a the live and not my playback time.

Do you have any idea abaout  this ?

Thanks

mpromonet, 2021-03-20:
Hi,

You may try with [openRTSP ](http://www.live555.com/openRTSP/)  specifying

```
-U <initial-absolute-seek-time> | request that the server seek to the specified absolute time (format: "YYYYMMDDTHHMMSSZ" or "YYYYMMDDTHHMMSS.<frac>Z") before streaming
-E <absolute-seek-end-time> | request that the server end streaming at the specified absolute time (format: "YYYYMMDDTHHMMSSZ" or "YYYYMMDDTHHMMSS.<frac>Z") (used only with -U<initial-absolute-seek-time>
```

Best Regards,
Michel.

MDalprato, 2023-05-31:
@ard-be did you find a way to do so ? 
I'm working on an Onvif project trying to get the edge recording from an NVR and I need to provide the 'range' header with the clock time in UTC format but I'm not able to archive that.
Any help ?

ard-be, 2023-06-12:
@MDalprato Sorry, we didn't work on that.

closed 2024-05-18 by mpromonet