mpromonet/webrtc-streamer #391: Segmentation Fault on Screencapture

prncvrm, 2021-03-17:
**Describe the bug**
```
[000:202][6663] (CapturerFactory.h:215): audiourl:screen://69734272 idx_audioDevice:-1/0
[000:202][6663] (PeerConnectionManager.cpp:1167): Cannot create capturer audio:screen://69734272
[000:203][775] (PeerConnectionManager.h:229): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.6382065118381824
[000:204][775] (PeerConnectionManager.h:235): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:0 peerid:0.6382065118381824
[1]    70366 segmentation fault  ./webrtc-streamer -u screen://0
```

Getting segmentation fault while trying to capture entire screen on macos

**To Reproduce**
Steps to reproduce the behavior:
To reproduce this, you'll need a mac os based binary, you can download from here (valid for next 24hours) https://free.keep.sh/9lxnXpQ8lD5z943m/webrtc-streamer
Just simply run the application, when you try to capture screen, sometimes you might get a screen(but that too a frozen one/Only the first frame) or get segmentation fault all together.

Works pretty fine with window capture though.

**Expected behavior**
Should be abe to capture screen as well.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: MacOS
 - Browser Any
 - Version v0.5.2-1-g15c8ac7-dirty/Darwin-x86_64

**Additional context**
Once this is fixed, we might have a macos compatible build on this repo as well.

cc : @mpromonet

prncvrm, 2021-03-19:
@mpromonet https://groups.google.com/g/discuss-webrtc/c/VsX5YrPmEmE i found this, can guide me here ?

prncvrm, 2021-03-19:
More debugging led me to this
```
[003:272][25347] (pacing_controller.cc:230): bwe:pacer_updated pacing_kbps=1928 padding_budget_kbps=0
[003:272][25347] (bitrate_prober.cc:114): Probe cluster (bitrate:min bytes:min packets): (3506086:6574:5)
[003:273][22019] (video_stream_encoder.cc:2051): OnBitrateUpdated, bitrate 1643660 stable bitrate = 190617 link allocation bitrate = 1643660 packet loss 0 rtt 0
[003:273][22019] (encoder_bitrate_adjuster.cc:225): Utilization factors for spatial index 0: link = 1.2, media = 1.2, wanted overshoot = 0 bps, available headroom = 660 bps, total utilization factor = 1.2
[003:273][22019] (video_stream_encoder.cc:1392): Adjusting allocation, fps = 210, from VideoBitrateAllocation [ [1643000] ], to VideoBitrateAllocation [ [1369167] ]
[003:315][5891] (video_stream_encoder.cc:1308): Incoming frame dropped due to that the encoder is blocked.
[003:316][5891] (video_stream_encoder.cc:1308): Incoming frame dropped due to that the encoder is blocked.
[003:344][25347] (video_stream_encoder.cc:1308): Incoming frame dropped due to that the encoder is blocked.
[003:345][25347] (video_stream_encoder.cc:1308): Incoming frame dropped due to that the encoder is blocked.
[003:345][25347] (video_stream_encoder.cc:1308): Incoming frame dropped due to that the encoder is blocked.
[003:346][25347] (video_stream_encoder.cc:1308): Incoming frame dropped due to that the encoder is blocked.
[003:347][25347] (video_stream_encoder.cc:1308): Incoming frame dropped due to that the encoder is blocked.
[003:374][25347] (probe_bitrate_estimator.cc:159): Probing successful [cluster id: 3] [send: 4888 bytes / 11 ms = 3554909 bps ] [receive: 5765 bytes / 13 ms = 3547692 bps]
[003:374][25347] (probe_bitrate_estimator.cc:159): Probing successful [cluster id: 3] [send: 6041 bytes / 14 ms = 3452 kbps ] [receive: 6918 bytes / 14 ms = 3953142 bps]
[003:374][25347] (probe_controller.cc:280): Measured bitrate: 3452000 Minimum to probe further: 2454260
[003:374][25347] (goog_cc_network_control.cc:688): bwe 389770646 pushback_target_bps=3452000 estimate_bps=3452000
[003:374][25347] (pacing_controller.cc:230): bwe:pacer_updated pacing_kbps=3797 padding_budget_kbps=0
[003:374][25347] (bitrate_prober.cc:114): Probe cluster (bitrate:min bytes:min packets): (5000000:9375:5)
[003:375][5891] (video_stream_encoder.cc:2051): OnBitrateUpdated, bitrate 2500000 stable bitrate = 95403 link allocation bitrate = 3242363 packet loss 0 rtt 0
[003:375][5891] (encoder_bitrate_adjuster.cc:225): Utilization factors for spatial index 0: link = 1.2, media = 1.2, wanted overshoot = 0 bps, available headroom = 742363 bps, total utilization factor = 1.2
[003:375][5891] (video_stream_encoder.cc:1392): Adjusting allocation, fps = 228, from VideoBitrateAllocation [ [2500000] ], to VideoBitrateAllocation [ [2083333] ]
[003:422][22019] (probe_bitrate_estimator.cc:159): Probing successful [cluster id: 4] [send: 6640 bytes / 9 ms = 5902222 bps ] [receive: 7453 bytes / 9 ms = 6624888 bps]
[003:422][22019] (probe_bitrate_estimator.cc:159): Probing successful [cluster id: 4] [send: 7729 bytes / 13 ms = 4756307 bps ] [receive: 8543 bytes / 13 ms = 5257230 bps]
[003:422][22019] (probe_bitrate_estimator.cc:159): Probing successful [cluster id: 4] [send: 8819 bytes / 14 ms = 5039428 bps ] [receive: 9633 bytes / 13 ms = 5928 kbps]
[003:422][22019] (goog_cc_network_control.cc:688): bwe 389770694 pushback_target_bps=5039428 estimate_bps=5039428
[003:422][22019] (pacing_controller.cc:230): bwe:pacer_updated pacing_kbps=5543 padding_budget_kbps=0
[003:423][5891] (video_stream_encoder.cc:2051): OnBitrateUpdated, bitrate 2500000 stable bitrate = 8120 link allocation bitrate = 4701400 packet loss 0 rtt 0
[003:423][5891] (encoder_bitrate_adjuster.cc:225): Utilization factors for spatial index 0: link = 1.2, media = 1.2, wanted overshoot = 0 bps, available headroom = 2201400 bps, total utilization factor = 1.2
[003:423][5891] (video_stream_encoder.cc:1392): Adjusting allocation, fps = 230, from VideoBitrateAllocation [ [2500000] ], to VideoBitrateAllocation [ [2083333] ]
[003:513][6407] (connection.cc:790): Conn[750d6600:0:Net[en0:192.168.2.x/24:Unknown:id=1]:HCCaxzl2:1:0:local:udp:192.168.2.x:50658->8gjapiys:1:1853817087:prflx:udp:192.168.2.x:64181|CRWS|S|1|0|7962083765675507198|3]: UpdateState(), ms since last received response=300, ms since last received data=37, rtt=100, pings_since_last_response=
[003:513][6407] (connection.cc:790): Conn[7701e800:0:Net[en0:192.168.2.x/24:Unknown:id=1]:HCCaxzl2:1:0:local:udp:192.168.2.x:50658->Sd0uwchw:1:1686052863:stun:udp:180.151.78.x:64181|C-xI|-|0|0|7241541910156688894|-]: UpdateState(), ms since last received response=389770785, ms since last received data=389770785, rtt=6000, pings_since_last_response=6c69636c556f76486a565675
[003:525][22019] (video_stream_encoder.cc:1308): Incoming frame dropped due to that the encoder is blocked.
[003:551][5891] (video_stream_encoder.cc:1308): Incoming frame dropped due to that the encoder is blocked.
[003:571][22019] (video_stream_encoder.cc:1308): Incoming frame dropped due to that the encoder is blocked.
[003:582][22019] (video_stream_encoder.cc:1308): Incoming frame dropped due to that the encoder is blocked.
[003:583][22019] (video_stream_encoder.cc:1308): Incoming frame dropped due to that the encoder is blocked.
[003:600][22019] (video_stream_encoder.cc:1308): Incoming frame dropped due to that the encoder is blocked.
[003:601][22019] (video_stream_encoder.cc:1308): Incoming frame dropped due to that the encoder is blocked.
```

the highlight here is 
```
Incoming frame dropped due to that the encoder is blocked.
```
cc : @mpromonet

prncvrm, 2021-03-19:
@mpromonet i resolved the seg fault btw,
now the issue is only first frame comes up and in log i get these two
Incoming frame dropped due to that the encoder is blocked.
```
Incoming frame dropped due to that the encoder is blocked.
..
..
..
[003:027][22019] (video_stream_encoder.cc:1542): Dropping frame. Too large for target bitrate.

```
Can you help here?

mpromonet, 2021-03-19:
Hi,

Recently I add for encoded data a little delay to reduce the situation "encoder is blocked" https://github.com/mpromonet/webrtc-streamer/commit/d48ebc589f2f4d27e15f550f36996b53e5c38008. This is clearly not a good solution, it should be better to wait for encoder availibility, but this need to dig a bit more in WebRTC SDK.

Best Regards,
Michel.

prncvrm, 2021-03-20:
@mpromonet thank you, but the https://groups.google.com/g/discuss-webrtc/c/VsX5YrPmEmE conversation helped me to figure the fix. Closing this. will submit a PR soon for entire mac build support as well :-)

closed 2021-03-20 by prncvrm

mpromonet, 2021-03-20:
Hi,

Maybe the fix for windows https://github.com/mpromonet/webrtc-streamer/commit/05ff828f4480a1f367a29fc64ed1f8329f1f5621 is not correct on mac ?

Best Regards,
Michel.

prncvrm, 2021-03-21:
Not really, the issue wasn't from decoder/encoder end