mpromonet/webrtc-streamer #392: RTSP stream not working (Amcrest)

MisterMelDev, 2021-03-18:
**Describe the bug**
When trying to use an RTSP stream from an Amcrest camera, the stream doesn't work and the following error appears in the console: https://pastebin.com/qYyV8QyN. The same url works fine when connecting with VLC.

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Browse: Google Chrome
 - Version: v0.5.2

mpromonet, 2021-03-20:
Hi,

The DESCRIBE answer 404 indicate the sever does not know the rtsp url.
Maybe you should url encode characters in your password ?

Best Regards,
Michel.

MisterMelDev, 2021-03-21:
After doing some troubleshooting, I've discovered the issue did have to do with URL encoding, just not the password. My password does contain special characters but I had already escaped those. What I discovered was that the '&' in the URL was causing issues. After replacing this with '%26', it works perfectly. For future reference, the URL you should use for Amcrest cameras is:

> rtsp://user:pass@ipaddr:554/cam/realmonitor?channel=1%26subtype=0

closed 2021-03-21 by MisterMelDev

shift16, 2025-08-03:
@melluh Your solution should be added to the FAQ as it fixed my issue as well!