mpromonet/webrtc-streamer #393: Streams with pred_weight_table unsupported

renjw1988, 2021-03-23:
I use this to play rtsp and the video seem to have many dropped frames. The same rtsp is play ok in VLC media player. And I see many log like "[557:515][5128] (h264_bitstream_parser.cc:206): Streams with pred_weight_table unsupported." from the webrtc-streamer. The rtsp is mp4(avc1) file stream from the ffserver.The mp4 file and rtsp can play ok in VLC, But in the WebRTC will dropped many frames,Whether use the webcomponent or not.

mpromonet, 2021-03-23:
Hi @renjw1988 

VLC use also live555 to get frames, so it should be about the same.
Did you tried to use RTP over TCP ? This is set by the options `rtptransport=tcp`

Best Regards,
Michel.

renjw1988, 2021-03-23:
> {
    "urls":{
	    "Test": {"video": "rtsp://127.0.0.1:554/testvideo","options":"rtptransport=tcp&timeout=60"}
    }
}

I use this options and still have the same problem.
VLC use also live555 ,but maybe something is different when webrtc-streamer decode the rtsp video?
In fact,I tryed use vlc to play this rtsp on my desktop and use webrtc to play "screen://0". The video(record from my screen) played well.

mpromonet, 2021-03-24:
Hi @renjw1988 

Then you may look to the log, enabling some verbosity level.

Best Regards,
Michel.

closed 2021-07-04 by mpromonet

mykolak-dev, 2021-08-06:
Same for me. Did you find how to fix it?