mpromonet/webrtc-streamer #394: Can't connect to webrtc-streamer in another machine

castilho25, 2021-03-26:
Hello mpromonet,

I'm launching a webrtc-streamer in a Jetson Nano Board (aarch64) with the docker command
`docker run -p 8090:8090 -it mpromonet/webrtc-streamer -H 0.0.0.0:8090 rtsp://<url_rtsp>`

when, in Jetson Nano,  i access via browser to <jetson-ip>:8090, the webrtc-streamer page is launched and i can watch the video streaming.
Otherwise, when I trying to connect via a laptop to <jetson-ip>:8090, the webrtc-streamer page is launched but I can't watch the video streaming.

The same happens, when i launch the webrtc-streamer in laptop and try to access via Jetson Nano.

Do you know why is that happening and what can I do to fix it?
Thank you :)

mpromonet, 2021-03-26:
Hi @castilho25 

Did you tried with `--net=host` ?
Usually this occurs because the published ICE candidate is the docker IP that is not reachable from your laptop.

Best Regards,
Michel.

sheryl-intertie, 2022-10-11:
Hi @mpromonet

I am using the docker run option --network=host and I am still seeing some errors and performance lag of the stream when not accessing the stream from the same laptop.  The errors start about 20 seconds after the request.  These errors do not appear when I access the stream from a browser on the same laptop as this webrtc-streamer docker container.  From the remote location, I do see the stream on the default html at port 8000.  Here are the errors:

1205 bytes to host 192.168.0.x:63377 (192.168.0.x:63377) failed with error 11
[683:732][11] (stun_port.cc:305): Port[1807d010:0:1:0:local:Net[wlp2s0:192.168.0.x/24:Unknown:id=9]]: UDP send of 1205 bytes to host 192.168.0.x:63377 (192.168.0.x:63377) failed with error 11
[684:922][11] (stun_port.cc:305): Port[1807d010:0:1:0:local:Net[wlp2s0:192.168.0.x/24:Unknown:id=9]]: UDP send of 70 bytes to host 192.168.0.x:63377 (192.168.0.x:63377) failed with error 11
[684:932][11] (stun_port.cc:305): Port[1807d010:0:1:0:local:Net[wlp2s0:192.168.0.x/24:Unknown:id=9]]: UDP send of 1109 bytes to host 192.168.0.x:63377 (192.168.0.x:63377) failed with error 11
[684:944][11] (stun_port.cc:305): Port[1807d010:0:1:0:local:Net[wlp2s0:192.168.0.x/24:Unknown:id=9]]: UDP send of 1110 bytes to host 192.168.0.x:63377 (192.168.0.x:63377) failed with error 11
[684:956][11] (stun_port.cc:305): Port[1807d010:0:1:0:local:Net[wlp2s0:192.168.0.x/24:Unknown:id=9]]: UDP send of 1110 bytes to host 192.168.0.x:63377 (192.168.0.x:63377) failed with error 11
[684:968][11] (stun_port.cc:305): Port[1807d010:0:1:0:local:Net[wlp2s0:192.168.0.x/24:Unknown:id=9]]: UDP send of 1110 bytes to host 192.168.0.x:63377 (192.168.0.x:63377) failed with error 11
[737:114][11] (stun_port.cc:305): Port[1807d010:0:1:0:local:Net[wlp2s0:192.168.0.x/24:Unknown:id=9]]: UDP send of 1206 bytes to host 192.168.0.x:63377 (192.168.0.x:63377) failed with error 11
[737:114][11] (stun_port.cc:305): Port[1807d010:0:1:0:local:Net[wlp2s0:192.168.0.x/24:Unknown:id=9]]: UDP send of 1206 bytes to host 192.168.0.x:63377 (192.168.0.x:63377) failed with error 11
[737:119][11] (stun_port.cc:305): Port[1807d010:0:1:0:local:Net[wlp2s0:192.168.0.x/24:Unknown:id=9]]: UDP send of 1206 bytes to host 192.168.0.x:63377 (192.168.0.x:63377) failed with error 11
[737:124][11] (stun_port.cc:305): Port[1807d010:0:1:0:local:Net[wlp2s0:192.168.0.x/24:Unknown:id=9]]: UDP send of 1206 bytes to host 192.168.0.x:63377 (192.168.0.x:63377) failed with error 11

I found that error 11 is EAGAIN (I think)
EAGAIN (...) The socket is marked nonblocking and the requested operation would block. (...)

EAGAIN (Internet domain datagram sockets) The socket referred to by sockfd had not previously been bound to an address and, upon attempting to bind it to an ephemeral port, it was determined that all port numbers in the ephemeral port range are currently in use. 

I am not sure how to fix this.  Do you know?  

Thank you in advance for your help with this.

Sheryl