mpromonet/webrtc-streamer #395: Install failed

asterissco, 2021-03-26:
**Describe the bug**
Install failed when try access to webrtc libs

**To Reproduce**
Follow the manual

**Additional context**

```
x live/liveMedia/include/HLSSegmenter.hh
-- Performing Test CXX_SUPPORTS_STATIC_LIBGCC
-- Performing Test CXX_SUPPORTS_STATIC_LIBGCC - Success
-- Performing Test CXX_SUPPORTS_STATIC_LIBSTDCPP
-- Performing Test CXX_SUPPORTS_STATIC_LIBSTDCPP - Success
-- Configuring done
CMake Error at CMakeLists.txt:131 (add_executable):
  Cannot find source file:

    /usr/src/webrtc/src/out/Release/obj/rtc_base/rtc_json/json.o

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx

CMake Error at CMakeLists.txt:131 (add_executable):
  No SOURCES given to target: webrtc-streamer

CMake Generate step failed.  Build files cannot be regenerated correctly.

```

mpromonet, 2021-03-26:
Hi,

You probably not get WebRTC SDK, or didnot give the path to allow cmake to reach it.

Best Regards,
Michel.

asterissco, 2021-03-26:
Hello!

I do this. This not get WebRTC SDK? I try to "find . -iname "json.o" " but not exists

```
pushd ..
git clone https://chromium.googlesource.com/chromium/tools/depot_tools.git
export PATH=$PATH:`realpath depot_tools`
popd

mkdir ../webrtc
pushd ../webrtc
fetch --no-history webrtc 
popd
```

mpromonet, 2021-03-26:
Hi,

Then probably the build failed...
You can rerun it running `rm -rf ../webrtc/src/out && cmake .`

Best Regards,
Michel.

asterissco, 2021-03-26:
Yes, failed.

I try to fix it. Thanks!
```

GTK_FOUND = 
ERROR at //build_overrides/build.gni:39:13: Script returned non-zero exit code.
  _result = exec_script("//build/mac/should_use_hermetic_xcode.py",
            ^----------
Current dir: /usr/src/webrtc/src/out/Release/
Command: python /usr/src/webrtc/src/build/mac/should_use_hermetic_xcode.py linux
Returned 1.
stderr:

Traceback (most recent call last):
  File "/usr/src/webrtc/src/build/mac/should_use_hermetic_xcode.py", line 28, in <module>
    import mac_toolchain
  File "/usr/src/webrtc/src/build/mac/../mac_toolchain.py", line 25, in <module>
    import pkg_resources
ImportError: No module named pkg_resources

See //build/toolchain/toolchain.gni:10:1: whence it was imported.
import("//build_overrides/build.gni")
^-----------------------------------
See //build/config/coverage/coverage.gni:5:1: whence it 
```

mpromonet, 2021-03-26:
Hi,

You need to install `python-pkg-resources`.
But if you try to build, there is no build for mac... it seems @prncvrm is working on this (see https://github.com/mpromonet/webrtc-streamer/issues/213)

Best Regards,
Michel.

closed 2021-03-26 by asterissco

asterissco, 2021-03-26:
I install this packet and "apt-get install libgtk-3-dev" and works fine!!

Thanks!

Garyguosha, 2021-03-31:
Hello  asterissco ：excuse me, I've compiled several times without success. Can you share with me how you compiled it?

asterissco, 2021-03-31:
With this works fine on Ubuntu 20:04 LXC Container

```
#### 1

apt-get install git build-essential vim net-tools python2.7 python-pkg-resources libgtk-3-dev

#### 2

git clone https://github.com/mpromonet/webrtc-streamer

#### 3 

pushd ..
git clone https://chromium.googlesource.com/chromium/tools/depot_tools.git
export PATH=$PATH:`realpath depot_tools`
popd

#### 4 

mkdir ../webrtc
pushd ../webrtc
fetch --no-history webrtc      
popd
    
#### 5 

cmake . && make 
```

Garyguosha, 2021-04-01:
Did you encounter such an error when compiling? I've already sent the depot_ Tools all files are converted to UINX format, but unfortunately still can't?

root@27370242ef06:/home/webrtc# fetch --no-history webrtc
/usr/bin/env: 'bash\r': No such file or directory
/home/guosha/depot_tools/bootstrap_python3: line 32: bootstrap-3.8.0.chromium.8_bin/python3/bin/python3: No such file or direct                            ory
cat: /home/guosha/depot_tools/python3_bin_reldir.txt: No such file or directory
/home/guosha/depot_tools/vpython3: line 52: /home/guosha/depot_tools/.cipd_bin/vpython3: No such file or directory

asterissco, 2021-04-01:
Try to compile with python2.7