mpromonet/webrtc-streamer #396: Audio Build issues on Mac

prncvrm, 2021-03-31:
**Describe the bug**
```
Undefined symbols for architecture x86_64:
  "webrtc::AudioDeviceModule::Create(webrtc::AudioDeviceModule::AudioLayer, webrtc::TaskQueueFactory*)", referenced from:
      PeerConnectionManager::PeerConnectionManager(std::__1::list<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, Json::Value const&, webrtc::AudioDeviceModule::AudioLayer, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, bool) in PeerConnectionManager.cpp.o
ld: symbol(s) not found for architecture x86_64
```
**To Reproduce**
Try building with `HAVE_SOUND` set in cmake on macos
**Expected behavior**
Should be built 

**Screenshots**
NA
**Desktop (please complete the following information):**
 - OS: MAC
 - Browser Any
 - Version latest

**Additional context**
@mpromonet can you help here or guide? the mac build is almost done.

mpromonet, 2021-03-31:
Hi,

I guess you build WebRTC SDK & webrtc-streamer in different situation. In actual https://github.com/mpromonet/webrtc-streamer/blob/master/CMakeLists.txt the HAVE_SOUND flag is set consistently with alsa availibility to build WebRTC SDK and webrtc-streamer.
Probably you should build WebRTC with `rtc_include_internal_audio_device=false` if no sound is available, defining HAVE_SOUND if SDK was build without audio support probably cause the problem you describe ?

Best Regards,
Michel.

prncvrm, 2021-03-31:
@mpromonet thanks for the quick reply. I'll try and let you know.

prncvrm, 2021-03-31:
@mpromonet building webrtc with `rtc_include_internal_audio_device=false` results in failure. the following is the log.
```
ubprocess.CalledProcessError: Command '['../../third_party/llvm-build/Release+Asserts/bin/clang++', '-B', '/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/', '-Wl,-fatal_warnings', '-stdlib=libc++', '-arch', 'x86_64', '-Werror', '-nostdlib++', '-isysroot', '../../../../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk', '-mmacosx-version-min=10.11.0', '-Wl,-ObjC', '-o', './webrtc_lib_link_test', '-Wl,-filelist,./webrtc_lib_link_test.rsp', '-framework', 'Foundation', '-framework', 'AppKit', '-framework', 'IOKit', '-framework', 'IOSurface', '-framework', 'AVFoundation', '-framework', 'CoreMedia']' returned non-zero exit status 1
[3198/3233] LINK ./webrtc_perf_tests
FAILED: webrtc_perf_tests
TOOL_VERSION=1615919258 ../../build/toolchain/apple/linker_driver.py -Wcrl,strippath,/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/strip ../../third_party/llvm-build/Release+Asserts/bin/clang++ -B /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/  -Wl,-fatal_warnings -stdlib=libc++ -arch x86_64 -Werror -nostdlib++ -isysroot ../../../../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk -mmacosx-version-min=10.11.0 -Wl,-ObjC -o "./webrtc_perf_tests" -Wl,-filelist,"./webrtc_perf_tests.rsp" -framework Foundation -framework AVFoundation -framework CoreMedia -framework VideoToolbox -framework CoreGraphics -framework CoreVideo -framework QuartzCore -framework OpenGL -framework Cocoa
Undefined symbols for architecture x86_64:
  "webrtc::AudioDeviceModule::Create(webrtc::AudioDeviceModule::AudioLayer, webrtc::TaskQueueFactory*)", referenced from:
      webrtc::VideoQualityTest::CreateAudioDevice() in video_quality_test.o
ld: symbol(s) not found for architecture x86_64
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
Traceback (most recent call last):
  File "../../build/toolchain/apple/linker_driver.py", line 305, in <module>
    Main(sys.argv)
  File "../../build/toolchain/apple/linker_driver.py", line 98, in Main
    subprocess.check_call(compiler_driver_args, env=env)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['../../third_party/llvm-build/Release+Asserts/bin/clang++', '-B', '/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/', '-Wl,-fatal_warnings', '-stdlib=libc++', '-arch', 'x86_64', '-Werror', '-nostdlib++', '-isysroot', '../../../../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk', '-mmacosx-version-min=10.11.0', '-Wl,-ObjC', '-o', './webrtc_perf_tests', '-Wl,-filelist,./webrtc_perf_tests.rsp', '-framework', 'Foundation', '-framework', 'AVFoundation', '-framework', 'CoreMedia', '-framework', 'VideoToolbox', '-framework', 'CoreGraphics', '-framework', 'CoreVideo', '-framework', 'QuartzCore', '-framework', 'OpenGL', '-framework', 'Cocoa']' returned non-zero exit status 1_
```
I rebuilt webrtc with `rtc_include_internal_audio_device=true` and tried building `webrtc-streamer` and got the same error as stated initially.

prncvrm, 2021-03-31:
just to update, i want to use recording devices on the mac machine.

mpromonet, 2021-04-02:
Hi,

Then maybe options to enable audio capture of WebRTC SDK on macos are different from Linux (alsa or pulse) & Windows(always enable).

Best Regards,
Michel.

prncvrm, 2021-04-02:
yes, looks like this might be the same. @mpromonet any idea or references from where i can refer this and get help?

mpromonet, 2021-04-02:
Hi,

If you have errors building the WebRTC SDK, you might report as en issue in https://bugs.chromium.org/p/webrtc/issues/list.

Best Regards,
Michel.

prncvrm, 2021-04-02:
no, i really don't think i've got issue with building webrtc sdk, its the audio thing on webrtc-streamer for mac os, Do you know how can i enable audio on webrtc-streamer for mac-oses? @mpromonet

mpromonet, 2021-04-02:
Hi,

In your previous post you post 
```
[3198/3233] LINK ./webrtc_perf_tests
FAILED: webrtc_perf_tests
TOOL_VERSION=1615919258 ../../build/toolchain/apple/linker_driver.py -Wcrl,strippath,/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/strip ../../third_party/llvm-build/Release+Asserts/bin/clang++ -B /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/  -Wl,-fatal_warnings -stdlib=libc++ -arch x86_64 -Werror -nostdlib++ -isysroot ../../../../../../../../Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk -mmacosx-version-min=10.11.0 -Wl,-ObjC -o "./webrtc_perf_tests" -Wl,-filelist,"./webrtc_perf_tests.rsp" -framework Foundation -framework AVFoundation -framework CoreMedia -framework VideoToolbox -framework CoreGraphics -framework CoreVideo -framework QuartzCore -framework OpenGL -framework Cocoa
Undefined symbols for architecture x86_64:
```

In webrtc-streamer, when HAVE_AUDIO is defined the audio capture use `webrtc::AudioDeviceModule::Create` (and then it need to exists in the WebRTC SDK you build). If HAVE_AUDIO is not defined, a fake capturer is used.

Best Regards,
Michel.

prncvrm, 2021-04-02:
yes, i've already done the same @mpromonet . I've built WebRTC SDK with AudioDeviceModule by enabling `rtc_include_internal_audio_device=true` flag. but it is still throwing me `Undefined symbols`

prncvrm, 2021-04-02:
@mpromonet more debugging lead me to conclusion that the definition argument list of a function `webrtc::AudioDeviceModule::Create` doesn't match with one used in webrtc-streamer. 
the error says 
```
Undefined symbols for architecture x86_64:
webrtc::AudioDeviceModule::Create(webrtc::AudioDeviceModule::AudioLayer, webrtc::TaskQueueFactory*)
...
...
```
The last parameter,  is a `webrtc::TaskQueueFactory*` and not `webrtc::TaskQueueFactory`

mpromonet, 2021-04-03:
Hi @prncvrm 

Anyway if the prototype is not correct, it will fail at compilation time, not at link time.
You may look to objects and symbols in webrtc libs (using nm, c++filt,...)
nm libwebrtc.a | c++filt | grep AudioDeviceModule::Create
...
0000000000000000 T webrtc::AudioDeviceModule::Create(webrtc::AudioDeviceModule::AudioLayer, webrtc::TaskQueueFactory*)
...

Best Regards,
Michel.

prncvrm, 2021-04-03:
hi @mpromonet , thank you for the quickest update. i just did look to objects and symbol by the commands you provided and here's the screenshot for the same.
[image]

If I ain't wrong, the build for WebRTC does contains the symbols. I m literally struck on this for days now. 😖 Anything else i could try with?

mpromonet, 2021-04-03:
Hi @prncvrm 

The symbol seems the one that is reported as unresolved, then I guess the library you are linking with is not the one that have the symbol.

Best Regards,
Michel.

prncvrm, 2021-04-03:
@mpromonet , i m linking the correct libraries. i just rebuilt the entire webrtc on my machine all over again, tested the built library with `nm` command, and then tried building webrtc-streamer again and got the same error. Is there a possibility we could connect over zoom/google hangout someday/sometime?

mpromonet, 2021-04-03:
Hi @prncvrm 

Maybe the link order, you could try to put libwebrtc add the end of link command...

Best Regards,
Michel.

prncvrm, 2021-04-03:
@mpromonet i m building this via cmake, didn't get it, could you please elaborate.

mpromonet, 2021-04-05:
Hi @prncvrm,

I was suggesting to look to the link command running `make VERBOSE=1` and try to link against `libwebrtc` at the end of the link command. However looking to the Cmakefile.txt, it seems after PeerConnectionManager.o. 

Maybe some tricks differ on MacOS. However it seems strange that this symbol is causing problem but not other, this is why my guess stay on a mix between different compilation of WebRTC & webrtc-streamer.....

Best Regards,
Michel.

prncvrm, 2021-04-09:
hey @mpromonet , Thank you, its finally resolved. there was some issue with the cmake file. 

thank you, closing this for now.

closed 2021-04-09 by prncvrm