mpromonet/webrtc-streamer #397: ubuntu16.0.4 编译出错

Garyguosha, 2021-03-31:
/usr/bin/env: ‘bash\r’: No such file or directory
/mnt/hgfs/linuxwebrtc/depot_tools/bootstrap_python3: line 32: bootstrap-3.8.0.chromium.8_bin/python3/bin/python3: No such file or directory
cat: /mnt/hgfs/linuxwebrtc/depot_tools/python3_bin_reldir.txt: No such file or directory
/mnt/hgfs/linuxwebrtc/depot_tools/vpython3: line 52: /mnt/hgfs/linuxwebrtc/depot_tools/.cipd_bin/vpython3: No such file or directory

mpromonet, 2021-03-31:
Hi,

May be you should try to explain your situation,  however it is more than probable than it doesnot build on ubuntu16.04.

Best Regards,
Michel.

closed 2021-03-31 by mpromonet

Garyguosha, 2021-04-01:
At the beginning, I was prompted that my compiled file format is doc, and I will depot it_ All tools files are converted to UINX file format, but the same error is reported

/usr/bin/env: 'bash\r': No such file or directory
/home/guosha/depot_tools/bootstrap_python3: line 32: bootstrap-3.8.0.chromium.8_bin/python3/bin/python3: No such file or direct                            ory
cat: /home/guosha/depot_tools/python3_bin_reldir.txt: No such file or directory
/home/guosha/depot_tools/vpython3: line 52: /home/guosha/depot_tools/.cipd_bin/vpython3: No such file or directory

mpromonet, 2021-04-01:
Hi,

This is probably some trouble with crlf git configuration on your platform, you should look to webrtc, maybe https://webrtc.github.io/webrtc-org/native-code/development/ or https://dev.chromium.org/developers/how-tos/depottools or ...

Best Regards,
Michel.