mpromonet/webrtc-streamer #398: Stream behind nat with local rtsp server

asterissco, 2021-03-31:
**Describe the bug**
Not a bug, just a question

**To Reproduce**

```
server-local # ./webrtc-streamer -s- -T0.0.0.0:3478 -tturn:turn@<public_static_ip>:3478  rtsp://<local_ip>:5540/channel-3

router# iptables -t nat -A PREROUTING -i ens8 -p tcp -m tcp --dport 3478 -j DNAT --to-destination server-local:3478
router# iptables -t nat -A PREROUTING -i ens8 -p udp -m udp --dport 3478 -j DNAT --to-destination server-local:3478
router# iptables -t nat-A PREROUTING -i ens8 -p tcp -m tcp --dport 8000 -j DNAT --to-destination server-local:8000
```

** Result ** 

```
[h264 @ 0x7fa640029300] no frame!
[002:118][15452] (h264_decoder_impl.cc:280): avcodec_send_packet error: -1094995529
[002:118][15452] (VideoDecoder.h:70): VideoDecoder::DecoderThread failure:-1
[h264 @ 0x7fa640029300] non-existing PPS 0 referenced
[h264 @ 0x7fa640029300] decode_slice_header error
[h264 @ 0x7fa640029300] no frame!
[002:158][15452] (h264_decoder_impl.cc:280): avcodec_send_packet error: -1094995529
[002:158][15452] (VideoDecoder.h:70): VideoDecoder::DecoderThread failure:-1
[h264 @ 0x7fa640029300] non-existing PPS 0 referenced
[h264 @ 0x7fa640029300] decode_slice_header error
[h264 @ 0x7fa640029300] no frame!
[002:219][15452] (h264_decoder_impl.cc:280): avcodec_send_packet error: -1094995529
[002:219][15452] (VideoDecoder.h:70): VideoDecoder::DecoderThread failure:-1
[h264 @ 0x7fa640029300] non-existing PPS 0 referenced
[h264 @ 0x7fa640029300] decode_slice_header error
[h264 @ 0x7fa640029300] no frame!
[002:248][15452] (h264_decoder_impl.cc:280): avcodec_send_packet error: -1094995529
[002:248][15452] (VideoDecoder.h:70): VideoDecoder::DecoderThread failure:-1
[h264 @ 0x7fa640029300] non-existing PPS 0 referenced
[h264 @ 0x7fa640029300] decode_slice_header error
[h264 @ 0x7fa640029300] no frame!
[002:278][15452] (h264_decoder_impl.cc:280): avcodec_send_packet error: -1094995529
[002:278][15452] (VideoDecoder.h:70): VideoDecoder::DecoderThread failure:-1
[h264 @ 0x7fa640029300] non-existing PPS 0 referenced
[h264 @ 0x7fa640029300] decode_slice_header error
[h264 @ 0x7fa640029300] no frame!
[002:348][15452] (h264_decoder_impl.cc:280): avcodec_send_packet error: -1094995529
[002:348][15452] (VideoDecoder.h:70): VideoDecoder::DecoderThread failure:-1
[h264 @ 0x7fa640029300] non-existing PPS 0 referenced
[h264 @ 0x7fa640029300] decode_slice_header error
[h264 @ 0x7fa640029300] no frame!
[002:378][15452] (h264_decoder_impl.cc:280): avcodec_send_packet error: -1094995529
[002:378][15452] (VideoDecoder.h:70): VideoDecoder::DecoderThread failure:-1
[h264 @ 0x7fa640029300] non-existing PPS 0 referenced
[h264 @ 0x7fa640029300] decode_slice_header error
[h264 @ 0x7fa640029300] no frame!
[002:408][15452] (h264_decoder_impl.cc:280): avcodec_send_packet error: -1094995529
[002:408][15452] (VideoDecoder.h:70): VideoDecoder::DecoderThread failure:-1
[h264 @ 0x7fa640029300] non-existing PPS 0 referenced
[h264 @ 0x7fa640029300] decode_slice_header error

```

Works fine on local but not on WAN, i am not sure this is possible.

Thanks for all! Regards!

asterissco, 2021-03-31:
Works fine on Chrome but crash with Firefox
```

Uncaught TypeError: recv.getParameters is not a function
    onicegatheringstatechange http://public_ip:8000/webrtcstreamer.js:188
    onicegatheringstatechange http://public_ip:8000/webrtcstreamer.js:186
webrtcstreamer.js:188:49
XHRPOSThttp://public_ip:8000/api/addIceCandidate?peerid=0.25277788688688907
```

asterissco, 2021-03-31:
Launch with 

./webrtc-streamer -s- -T0.0.0.0:3478 -tturn:turn@public_ip:3478

Stack errors (3) with Firefox

```
XHRPOSThttp://public_ip:8000/api/addIceCandidate?peerid=0.46153458840011974
[HTTP/1.1 500 OK 20ms]
Internal Server Error

Uncaught TypeError: recv.getParameters is not a function
    onicegatheringstatechange http://public_ip:8000/webrtcstreamer.js:188
    onicegatheringstatechange http://public_ip:8000/webrtcstreamer.js:186
webrtcstreamer.js:188:49

XHRPOSThttp://public_ip:8000/api/addIceCandidate?peerid=0.46153458840011974
[HTTP/1.1 500 OK 7ms]
Internal Server Error

WebRTC: ICE failed, add a STUN server and see about:webrtc for more details

```

sweetyguazi, 2022-05-13:
I met the same, have you solved it?