mpromonet/webrtc-streamer #399: An error occurred when using external network access

UkVOSklF, 2021-04-01:


UkVOSklF, 2021-04-01:
[014:994][2280] (PeerConnectionManager.h:229): OnRenegotiationNeeded peerid:0.07483307783068982
[014:997][2280] (PeerConnectionManager.h:235): OnSignalingChange state:0 peerid:0.07483307783068982
[014:998][12448] (video_send_stream_impl.cc:266): ERROR: Initial encoder max bitrate = -1 which is <= 0!
Start playing sink for "video/H264" subsession
markerSize:4
[027:694][15092] (PeerConnectionManager.cpp:776): Remove PeerConnection peerid:0.16177755425236984
[027:694][15092] (PeerConnectionManager.cpp:793): hangUp stream is no more used HDWebCam||
[027:695][15092] (PeerConnectionManager.cpp:801): hangUp stream closed HDWebCam||

#
# Fatal error in: ../../modules/video_capture/windows/sink_filter_ds.cc, line 820
# last system error: 0
# Check failed: (&main_checker_)->IsCurrent()

mpromonet, 2021-04-01:
Hi,

I guess from the lots of information in your description that you are trying Debug version for windows.
Did you try the release version ?

Best Regards,
Michel.

UkVOSklF, 2021-04-01:
# Fatal error in: ../../modules/video_capture/windows/sink_filter_ds.cc, line 820
# last system error: 0
# Check failed: (&main_checker_)->IsCurrent()

Mistakes like this can occur

mpromonet, 2021-04-01:
Hi,

I don't understand, but feel free to submit a Pull Request to fix this, I will review it.

Best Regards,
Michel.

closed 2021-04-13 by UkVOSklF