mpromonet/webrtc-streamer #401: CPU 100% problem makes the service unavailable

UkVOSklF, 2021-04-05:
After using Windwos to build a service
When playing 9 videos or multiple videos on the same page at the same time
The CPU load will reach 100%, causing the service to be unavailable
The console information is shown in the figure below

**Console information**

[024:401][9356] (PeerConnectionManager.h:144): OnStateChange channel:ClientDataChannel state:open
[024:402][9356] (PeerConnectionManager.h:151): OnMessage channel:ClientDataChannel msg:local channel openned
[024:404][9356] (PeerConnectionManager.h:151): OnMessage channel:ServerDataChannel msg:remote channel openned
[024:769][9356] (PeerConnectionManager.h:144): OnStateChange channel:ServerDataChannel state:open
[024:773][9356] (PeerConnectionManager.h:225): OnDataChannel
[024:774][9356] (PeerConnectionManager.h:144): OnStateChange channel:ClientDataChannel state:open
[024:775][9356] (PeerConnectionManager.h:151): OnMessage channel:ClientDataChannel msg:local channel openned
[024:777][9356] (PeerConnectionManager.h:151): OnMessage channel:ServerDataChannel msg:remote channel openned
[024:994][9356] (PeerConnectionManager.h:144): OnStateChange channel:ServerDataChannel state:open
[024:997][9356] (PeerConnectionManager.h:225): OnDataChannel
[024:997][9356] (PeerConnectionManager.h:144): OnStateChange channel:ClientDataChannel state:open
[024:998][9356] (PeerConnectionManager.h:151): OnMessage channel:ClientDataChannel msg:local channel openned
[025:003][9356] (PeerConnectionManager.h:151): OnMessage channel:ServerDataChannel msg:remote channel openned
[025:161][9356] (PeerConnectionManager.h:229): OnRenegotiationNeeded peerid:0.6552515570152369
[025:163][9356] (PeerConnectionManager.h:235): OnSignalingChange state:3 peerid:0.6552515570152369
[025:164][9356] (webrtc_video_engine.cc:3351): Absent receive stream; ignoring clearing encoded frame sink for ssrc 0
[025:164][9356] (PeerConnectionManager.h:214): OnAddStream nb video tracks:1
[025:165][9356] (PeerConnectionManager.h:229): OnRenegotiationNeeded peerid:0.6552515570152369
[025:167][9356] (PeerConnectionManager.h:235): OnSignalingChange state:0 peerid:0.6552515570152369
[025:322][9356] (PeerConnectionManager.h:144): OnStateChange channel:ServerDataChannel state:open
[025:325][9356] (PeerConnectionManager.h:225): OnDataChannel
[025:325][9356] (PeerConnectionManager.h:144): OnStateChange channel:ClientDataChannel state:open
[025:326][9356] (PeerConnectionManager.h:151): OnMessage channel:ClientDataChannel msg:local channel openned
[025:328][9356] (PeerConnectionManager.h:151): OnMessage channel:ServerDataChannel msg:remote channel openned
[025:699][9356] (PeerConnectionManager.h:144): OnStateChange channel:ServerDataChannel state:open
[025:707][9356] (PeerConnectionManager.h:225): OnDataChannel
[025:707][9356] (PeerConnectionManager.h:144): OnStateChange channel:ClientDataChannel state:open
[025:708][9356] (PeerConnectionManager.h:151): OnMessage channel:ClientDataChannel msg:local channel openned
[025:713][9356] (PeerConnectionManager.h:151): OnMessage channel:ServerDataChannel msg:remote channel openned
[027:279][13624] (PeerConnectionManager.cpp:776): Remove PeerConnection peerid:0.14257435401365903
[034:240][9356] (PeerConnectionManager.h:229): OnRenegotiationNeeded peerid:0.14257435401365903

mpromonet, 2021-04-05:
Hi UkVOSklF,

You my try the `-o` option that reduce compressing stuff but reduce format adaptation between RTSP stream and WebRTC stream.

Best Regards,
Michel.

Garyguosha, 2021-04-27:
hi
How to configure the -o parameter in docker compose,

[image]

mpromonet, 2021-04-27:
Hi Garyguosha,

May be using https://docs.docker.com/compose/compose-file/compose-file-v3/#command ?

Best Regards,
Michel.

Garyguosha, 2021-04-28:
hi mpromonet

I'm really sorry, but my configuration has been wrong. Can you configure an example.

Garyguosha, 2021-04-28:
hi UkVOSklF

Have you solved it。

mpromonet, 2021-04-28:
Hi @Garyguosha 

I think if you read the docker-compose documentation and make some tries, you will easily find the syntax to add `-o` to the command list. This is not specific to this project and there is plenty of example on internet.

Best Regards,
Michel.

Garyguosha, 2021-04-28:
hi
Download the depot in this way_ Tools, Download error.

[image]

mpromonet, 2021-04-28:
Hi Garyguosha,

Please could you reduce spoiling issues that have nothing to see your problem. 
Mixing runtime subject with build one.... what is the link.... `-o` is a runtime option that enable NullCodec, nothing to deal with build args of docker image.... it is hard to follow what does you mean...

Best Regards,
Michel.

closed 2021-05-24 by UkVOSklF