mpromonet/webrtc-streamer #404: Hangup on failed ice connection state

pickledgator, 2021-04-07:
**Describe the bug**
I'm having an issue where an unexpected hangup is called due the an ice connection state observation being `failed`. This is happening after the first ice candidate is sent from a remote peer to the server. Then all subsequent ice candidates return 500 Internal Server Error, since the peerConnection has closed. The problem is that it seems to be intermittent, but when the failure does happen, it happens after the first ice candidate is received.

Running the server with:
```
./webrtc-streamer -v -H 8000 -n axis -u rtsp://192.168.7.157/axis-media/media.3gp
```

**Work around**
If I comment out the following lines, my connections are established reliably every time.
https://github.com/mpromonet/webrtc-streamer/blob/master/inc/PeerConnectionManager.h#L239-L248

**To Reproduce**
Steps to reproduce the behavior:
* Send offer with /api/call
* Handle answer and set remote description on client side
* Start sending ice candidates via /api/addIceCandidate

**Desktop:**
 - OS: Ubuntu 18.04

**Smartphone:**
 - Device: Essential PH1
 - OS: Android 10
 - Flutter webrtc

Here is an output of the server's log:
```
[027:098][18627] (HttpServerRequestHandler.cpp:61): uri:/api/call
[027:098][18627] (PeerConnectionManager.cpp:650): call video:axis audio: options:
[027:098][18627] (PeerConnectionManager.cpp:987): CreatePeerConnectionCreatePeerConnection webrtcPortRange:0:65535
[027:098][18627] (PeerConnectionManager.cpp:989): CreatePeerConnectionCreatePeerConnection peerid:18636ccef0a8b495
[027:098][18627] (PeerConnectionManager.h:170): PeerConnectionObserverCreatePeerConnection peerid:18636ccef0a8b495
[027:098][18575] (peer_connection_factory.cc:357): Using default network controller factory
[027:098][18575] (bitrate_prober.cc:72): Bandwidth probing enabled, set to inactive
[027:099][18575] (pacing_controller.cc:230): bwe:pacer_updated pacing_kbps=300 padding_budget_kbps=0
[027:099][18575] (aimd_rate_control.cc:113): Using aimd rate control with back off factor 0.85
[027:099][18575] (remote_bitrate_estimator_single_stream.cc:72): RemoteBitrateEstimatorSingleStream: Instantiating.
[027:099][18575] (remote_estimator_proxy.cc:50): Maximum interval between transport feedback RTCP messages (ms): 250
[027:100][18575] (webrtc_session_description_factory.cc:185): DTLS-SRTP enabled; sending DTLS identity request (key type: 1).
[027:100][18576] (openssl_key_pair.cc:38): Making key pair
[027:100][18627] (PeerConnectionManager.h:177): PeerConnectionObserverCreateDataChannel peerid:18636ccef0a8b495
[027:100][18576] (openssl_key_pair.cc:91): Returning key pair
[027:100][18576] (boringssl_certificate.cc:187): Making certificate for WebRTC
[027:100][18575] (peer_connection.cc:2078): Local and Remote descriptions must be applied to get the SSL Role of the SCTP transport.
[027:100][18575] (PeerConnectionManager.h:229): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:18636ccef0a8b495
[027:101][18576] (boringssl_certificate.cc:243): Returning certificate
[027:101][18575] (webrtc_session_description_factory.cc:491): Setting new certificate.
[027:101][18627] (PeerConnectionManager.cpp:676): nbStreams local:0 remote:0 localDescription:0
[027:102][18576] (p2p_transport_channel.cc:532): Set backup connection ping interval to 25000 milliseconds.
[027:102][18576] (p2p_transport_channel.cc:541): Set ICE receiving timeout to 2500 milliseconds
[027:102][18576] (p2p_transport_channel.cc:548): Set ping most likely connection to 0
[027:102][18576] (p2p_transport_channel.cc:555): Set stable_writable_connection_ping_interval to 2500
[027:102][18576] (p2p_transport_channel.cc:568): Set presume writable when fully relayed to 0
[027:102][18576] (p2p_transport_channel.cc:586): Set regather_on_failed_networks_interval to 300000
[027:102][18576] (p2p_transport_channel.cc:593): Set receiving_switching_delay to 1000
[027:102][18576] (jsep_transport_controller.cc:990): Creating DtlsSrtpTransport.
[027:102][18576] (dtls_srtp_transport.cc:62): Setting RTCP Transport on video transport 0
[027:102][18576] (dtls_srtp_transport.cc:67): Setting RTP Transport on video transport a8003180
[027:102][18576] (p2p_transport_channel.cc:532): Set backup connection ping interval to 25000 milliseconds.
[027:102][18576] (p2p_transport_channel.cc:541): Set ICE receiving timeout to 2500 milliseconds
[027:102][18576] (p2p_transport_channel.cc:548): Set ping most likely connection to 0
[027:102][18576] (p2p_transport_channel.cc:555): Set stable_writable_connection_ping_interval to 2500
[027:102][18576] (p2p_transport_channel.cc:568): Set presume writable when fully relayed to 0
[027:102][18576] (p2p_transport_channel.cc:586): Set regather_on_failed_networks_interval to 300000
[027:102][18576] (p2p_transport_channel.cc:593): Set receiving_switching_delay to 1000
[027:102][18576] (jsep_transport_controller.cc:990): Creating DtlsSrtpTransport.
[027:102][18576] (dtls_srtp_transport.cc:62): Setting RTCP Transport on data transport 0
[027:102][18576] (dtls_srtp_transport.cc:67): Setting RTP Transport on data transport a80060f0
[027:102][18576] (p2p_transport_channel.cc:477): Received remote ICE parameters: ufrag=g6Og, renomination enabled
[... 437 lines cut ...]
[027:265][18761] (PeerConnectionManager.cpp:820): hangUp p�[027:399][18619] (HttpServerRequestHandler.cpp:61): uri:/api/addIceCandidate
[027:401][18619] (PeerConnectionManager.cpp:498): PEERID 18636ccef0a8b495
[027:401][18619] (PeerConnectionManager.cpp:518): ANSWER null
[027:401][18619] (HttpServerRequestHandler.cpp:82): code:500 answer:Internal Server Error
[027:550][18616] (HttpServerRequestHandler.cpp:61): uri:/api/addIceCandidate
[027:551][18616] (PeerConnectionManager.cpp:498): PEERID 18636ccef0a8b495
[027:551][18616] (PeerConnectionManager.cpp:518): ANSWER null
[027:551][18616] (HttpServerRequestHandler.cpp:82): code:500 answer:Internal Server Error
[027:706][18615] (HttpServerRequestHandler.cpp:61): uri:/api/addIceCandidate
[027:708][18615] (PeerConnectionManager.cpp:498): PEERID 18636ccef0a8b495
[027:708][18615] (PeerConnectionManager.cpp:518): ANSWER null
[027:708][18615] (HttpServerRequestHandler.cpp:82): code:500 answer:Internal Server Error
[027:867][18614] (HttpServerRequestHandler.cpp:61): uri:/api/addIceCandidate
[027:869][18614] (PeerConnectionManager.cpp:498): PEERID 18636ccef0a8b495
[027:869][18614] (PeerConnectionManager.cpp:518): ANSWER null
[027:869][18614] (HttpServerRequestHandler.cpp:82): code:500 answer:Internal Server Error
[028:029][18612] (HttpServerRequestHandler.cpp:61): uri:/api/addIceCandidate
[028:029][18612] (PeerConnectionManager.cpp:498): PEERID 18636ccef0a8b495
[028:030][18612] (PeerConnectionManager.cpp:518): ANSWER null
[028:030][18612] (HttpServerRequestHandler.cpp:82): code:500 answer:Internal Server Error
[028:183][18613] (HttpServerRequestHandler.cpp:61): uri:/api/addIceCandidate
[028:186][18613] (PeerConnectionManager.cpp:498): PEERID 18636ccef0a8b495
[028:186][18613] (PeerConnectionManager.cpp:518): ANSWER null
[028:186][18613] (HttpServerRequestHandler.cpp:82): code:500 answer:Internal Server Error
```

mpromonet, 2021-04-08:
Hi pickledgator,

Based on documentation https://hpbn.co/webrtc/ the connection should not be closed in case of state=failed.
However hangup should be called when ice connection is closed, otherwise the WebRTC stream will not be closed when client is no more present.

Best Regards,
Michel.

pickledgator, 2021-04-08:
That makes sense. Removing the state=failed check in that if statement resolves the issue I was having as well.

mpromonet, 2021-04-09:
Hi,

I made some tests but keeping only state=closed in the test doesnot detect if the webrtc consumer disappear. Probably need to understand more closely your problem.

Best Regards,
Michel.

closed 2024-05-18 by mpromonet