mpromonet/webrtc-streamer #405: Using docker to throw an error, unable to monitor the video

Garyguosha, 2021-04-07:
root@lambda-70DGA014CN:/home/guosha# docker run -p 8000:8000 -it mpromonet/webrtc-streamer -n axis  -u rtsp://admin:admin@192.168.1.137/11
Version:v0.6.1-1-gc83fe2f/Linux-x86_64 civetweb@v1.13 webrtc@03bce3f49d-dirty live555helper@897450c
{
        "urls" :
        {
                "axis" :
                {
                        "video" : "rtsp://admin:admin@192.168.1.137/11"
                }
        }
}Logger level:3
[000:000][1] (audio_device_pulse_linux.cc:1610): failed to connect context, error=-1
[000:000][1] (audio_device_pulse_linux.cc:144): failed to initialize PulseAudio
[000:000][1] (audio_device_impl.cc:342): Audio device initialization failed.
HTTP Listen at 0.0.0.0:8000
[004:977][11] (shared_x_display.cc:38): Unable to open display
[004:977][11] (shared_x_display.cc:38): Unable to open display
[005:097][1] (PeerConnectionManager.h:229): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.21694305718624274
[005:100][1] (PeerConnectionManager.h:235): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:3 peerid:0.21694305718624274
[005:101][1] (webrtc_video_engine.cc:3394): Absent receive stream; ignoring clearing encoded frame sink for ssrc 0
[005:101][1] (PeerConnectionManager.h:214): virtual void PeerConnectionManager::PeerConnectionObserver::OnAddStream(rtc::scoped_refptr<webrtc::MediaStreamInterface>) nb video tracks:1
Created new TCP socket 15 for connection
[005:102][12] (audio_device_pulse_linux.cc:1610): failed to connect context, error=-1
[005:102][12] (audio_device_pulse_linux.cc:144): failed to initialize PulseAudio
[005:103][12] (audio_device_impl.cc:342): Audio device initialization failed.
[005:103][12] (CapturerFactory.h:246): audiourl: idx_audioDevice:-1/-1
[005:103][12] (PeerConnectionManager.cpp:1167): Cannot create capturer audio:axis
[005:103][1] (PeerConnectionManager.h:229): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.21694305718624274
[005:105][1] (PeerConnectionManager.h:235): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:0 peerid:0.21694305718624274
[005:330][67] (livevideosource.h:175): LiveVideoSource:onData no decoder
[005:330][67] (livevideosource.h:175): LiveVideoSource:onData no decoder
Start playing sink for "video/H264" subsession
markerSize:4
[020:238][85] (PeerConnectionManager.cpp:776): Remove PeerConnection peerid:0.21694305718624274
[020:238][85] (PeerConnectionManager.cpp:793): hangUp stream is no more used axis||tcp60
[020:238][85] (PeerConnectionManager.cpp:801): hangUp stream closed axis||tcp60
Close session: video/H264
[020:242][1] (PeerConnectionManager.h:229): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.21694305718624274
[020:243][1] (PeerConnectionManager.h:235): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:5 peerid:0.21694305718624274

mpromonet, 2021-04-07:
Please read https://github.com/mpromonet/webrtc-streamer/blob/master/.github/CONTRIBUTING.md

closed 2021-04-07 by mpromonet

mpromonet, 2021-04-08:
Hi Garyguosha,

Please do not post again the same post in other issue, this is meaningless in the other thread.
You should better explain your problem, its context, what you tried, and may submit some PR to fixit.

Best Regards,
Michel.