mpromonet/webrtc-streamer #406: Cannot stream over HTTPS

castilho25, 2021-04-07:
Hey mpromonet,

I'm using your webrtc-streamer application for a while. I used the unsecured version over HTTP to connect the videos. Now, I'm trying to improve the security of my website, so, I'm trying to turn on the connections over HTTPS. I introduce the PEM file with 'BEGIN CERITIFICATE' and 'BEGIN PRIVATE KEY' but i think the application is not using this, because the connection is still by HTTP. 

What i'm doing wrong?

The commands I'm using are
`docker run -H 0.0.0.0:8090 -c ssl/certificates/se-certs.pem --network host -it mpromonet/webrtc-streamer`

I tried also with x86 exe and it didn't work too.

mpromonet, 2021-04-07:
Hi,

You probably need to put arguments (-H and -c ) after the image name and and -v to mount your pem file.

Best Regards,
Michel.

Garyguosha, 2021-04-08:
Hi,mpromonet
I use docker to run the program。

The commands I'm using are
` docker run -p 8000:8000 -it mpromonet/webrtc-streamer -n axis  -u rtsp://admin:admin@192.168.1.137/11`

The page can be accessed, but the video monitor cannot be opened，Please also see what it is，thank you.

root@lambda-70DGA014CN:/home/guosha# docker run -p 8000:8000 -it mpromonet/webrtc-streamer -n axis  -u rtsp://admin:admin@192.168.1.137/11
Version:v0.6.1-1-gc83fe2f/Linux-x86_64 civetweb@v1.13 webrtc@03bce3f49d-dirty live555helper@897450c
{
        "urls" :
        {
                "axis" :
                {
                        "video" : "rtsp://admin:admin@192.168.1.137/11"
                }
        }
}Logger level:3
[000:000][1] (audio_device_pulse_linux.cc:1610): failed to connect context, error=-1
[000:000][1] (audio_device_pulse_linux.cc:144): failed to initialize PulseAudio
[000:000][1] (audio_device_impl.cc:342): Audio device initialization failed.
HTTP Listen at 0.0.0.0:8000
[004:977][11] (shared_x_display.cc:38): Unable to open display
[004:977][11] (shared_x_display.cc:38): Unable to open display
[005:097][1] (PeerConnectionManager.h:229): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.21694305718624274
[005:100][1] (PeerConnectionManager.h:235): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:3 peerid:0.21694305718624274
[005:101][1] (webrtc_video_engine.cc:3394): Absent receive stream; ignoring clearing encoded frame sink for ssrc 0
[005:101][1] (PeerConnectionManager.h:214): virtual void PeerConnectionManager::PeerConnectionObserver::OnAddStream(rtc::scoped_refptr<webrtc::MediaStreamInterface>) nb video tracks:1
Created new TCP socket 15 for connection
[005:102][12] (audio_device_pulse_linux.cc:1610): failed to connect context, error=-1
[005:102][12] (audio_device_pulse_linux.cc:144): failed to initialize PulseAudio
[005:103][12] (audio_device_impl.cc:342): Audio device initialization failed.
[005:103][12] (CapturerFactory.h:246): audiourl: idx_audioDevice:-1/-1
[005:103][12] (PeerConnectionManager.cpp:1167): Cannot create capturer audio:axis
[005:103][1] (PeerConnectionManager.h:229): virtual void PeerConnectionManager::PeerConnectionObserver::OnRenegotiationNeeded() peerid:0.21694305718624274
[005:105][1] (PeerConnectionManager.h:235): virtual void PeerConnectionManager::PeerConnectionObserver::OnSignalingChange(webrtc::PeerConnectionInterface::SignalingState) state:0 peerid:0.21694305718624274
[005:330][67] (livevideosource.h:175): LiveVideoSource:onData no decoder
[005:330][67] (livevideosource.h:175): LiveVideoSource:onData no decoder
Start playing sink for "video/H264" subsession
markerSize:4

Garyguosha, 2021-04-08:
Hi castilho25,
When you encounter a local virtual machine in use, you can access the video, but other computers can display the page when they visit, but they can't load the video.
[image]

harpyeagle, 2021-04-08:
Hi,

In case this helps,
My solution for this is to run the webrtc-streamer docker behind a reverse proxy, this proxy is exposed to the world on 80/443 and protected using ( your local firewall rules) and if using a domain, by cloudflare or similar.

My setup is >
Docker host machine (unraid or similar)
Docker for Reverse proxy (Nginx Proxy Manager)
Router with firewall and external traffic on 80 / 443 port forwarded to proxy
Webrtc-streamer docker to stream my local nestcam rtsp streams. docker running in network host mode
Cloudflare ( as I use a domain with Lets encrypt )

Quick thanks to mpromonet for both webrtc-streamer and v4l2rtspserver - Im using them for nestbox cameras. Would not have been possible without these.
John

Garyguosha, 2021-04-09:
Do you mean this?

[image]

harpyeagle, 2021-04-09:
Hi Garyguosha, 

Not really...

This is a big topic, not directly related to webrtc-streamer, Im happy to help, but @mpromonet  let us know if you are happy for this off-topic thread to continue.

If I understand you correctly you want to access your stream, securely over https, outside your home network. **Exposing your home network to the world, is a big deal so I would suggest that you thread carefully and understand each step**. Saying that running your own home server is a great way to learn.

To allow the public access to the webrtc-streamer docker, you will be exposing ports on your local network to the outside world. Nginx proxy manager allows you to map the internal ports to the ports you have exposed.

In my case the ports I have exposed to the world are 80/443 (http/https). So when a visitor from the outside world visits my public IP, they are directed to the internal private IP of the nginx proxy manager docker, and in turn, the nginx proxy manager, resolves this request to the webrtc-streamer docker port.

For added points :) it would be good to have a domain associated with your pubic IP.

So the diagram is closer to;

> Public visitor go to your domain https://www.yourdomain.com
> https://www.yourdomain.com is resolved to your public ip - 92.123.45.7 
> Your router allows the https request (port 443) through via firewall rules and port forwarding to the Nginx proxy manager docker on e.g. 192.168.2.16:7818
> Nginx proxy manager performs a reverse proxy to webrtc-streamer docker on 192.168.2.15:8000
> The stream is displayed to the visitor on https://www.yourdomain.com

A lot to unpack, but searching on each of the topics above should give you a good foundation.

John

castilho25, 2021-04-09:
Hey harpyeagle, thank you for your contribuition! :)

There is a way to the webstream not run in network host mode?  Like creating a proxy network between nginx and the webstream. I tried and I can access to the webstreamer server but can't connect to the video. I think is because of the IceServer

harpyeagle, 2021-04-09:
As discussed here - it seems to be related to host / client discovery and how docker adds a layer of complexity  to this; https://github.com/node-webrtc/node-webrtc-examples/issues/2#issuecomment-526150719

I use the "host" network option, but let me spin up another instance in bridge mode and see what happens

harpyeagle, 2021-04-12:
@castilho25 I was not able to get this working in Bridge mode. 

As mentioned it relates to NAT and docker isolation. I did see some options for declaring the external IP from the docker but it would involve a change to the docker so this could be read in at run time .

One option that did work was to use the macvlan driver to give the docker its own IP on the network but I'm not sure if this brings any additional benefit over the network host option  ( other than allowing more ports, as the host IP is not being used)

John

harpyeagle, 2022-03-14:
@elliott3 I will help if I can. 

Would you consider using a docker ( or two ) instead?

Can I ask what the goal is, are you intending to give remote (external) access to the windows machine?

elliott3, 2022-03-14:
Hi @harpyeagle ,

Thank you for your help. My email is elli3ott@gmail.com. Our project needs to run on virtual machines. Can you email me about how to solve it? I could post the solution once I manage to stream my stuff.

mpromonet, 2022-03-14:
Hi @elliott3 

Please do not answer to an issue that is really different than yours. This is confusing and mixing subjects. The problem discuss here is on linux docker, then running on windows is not related.
By the way, increasing loglevel, or attached in a debugger may help you to answer the problem.

Best Regards,
Michel.

elliott3, 2022-03-15:
Hi Michael, 

Thank you for your update. I installed openssl and increased the log level. It still stops working without any reason.
Am I doing wrong?
[image]

All the best
Elliott

mpromonet, 2022-03-16:
Hi,

Probably nothing is wrong, and there is surely a reason to stop, it may be a non compatible version of openssl.  
You should look to the reason, maybe using a debugger, Dr Watson, or another way.

Best Regards,
Michel.