mpromonet/webrtc-streamer #407: Non local access issues

Garyguosha, 2021-04-09:
Hey mpromonet,

I use docker to run webrtc_ Streamer, can display video monitoring in local access, but only the page can be displayed when other computers access, and the video monitoring screen fails to display.
[image]

mpromonet, 2021-04-09:
Hi @Garyguosha 

Did you try running docker with `--net=host` ?

I guess the problem is because ICE negociation doesnot have candidate with the docker host, it has probably candidate with internal docker ip (that is not reachable outside docker) and candidate with your public ip, that is also not the docker host one.
A possible way without `--net=host` may be to run a STUN/TURN server on your LAN instead of using th google STUN.

Best Regards,
Michel.

Garyguosha, 2021-04-09:
ok  thanks

closed 2021-05-24 by mpromonet