mpromonet/webrtc-streamer #408: Jitsi connection problem

ginolaku, 2021-04-13:
we want to use our local RTSP streams to Jitsi meet ,  rtsp stream can be shown in localhost:8000
try to use XMPPVideoRoom.html. https://meet.jit.si/XXXXXroom cannot see the rtsp streaming

even browsing online demo
https://webrtc-streamer.herokuapp.com/xmppvideoroom.html
the situation is the same. 

[image]

mpromonet, 2021-04-13:
Hi @ginolaku,

It seems to not work anymore, it was always quite fragile, connection to jitsi, strophe & co need some work.... it seems more stable with janus-gateway.

Best Regards,
Michel.

mpromonet, 2021-07-17:
I close this issue, because since the online demo is running with `-o` it stream only H264, and jitsi by default use VP8, then something need to be change to have a working situation.

closed 2021-07-17 by mpromonet