mpromonet/webrtc-streamer #409: no audio when connecting through ssl

Coennie, 2021-04-13:
Hi There,

I'm running the version 6.2 on windows. When starting the stream without ssl video and audio works. When starting the stream through ssl video works, but I do not get any audio.

Does anyone else have this problem, and how can I debug this?

Regards

mpromonet, 2021-04-13:
Hi @Coennie 

This is strange that it could have a link between using https or http, maybe the problem is elsewhere.
Looking to the log, increase the verbosity could help you to collect more information.

Best Regards,
Michel.

Coennie, 2021-04-13:
I've enabled the verbose, but can't figure out what is wrong. I've attached the log is this something you can help with?

https://drive.google.com/file/d/1kktHg3ogz4emCv6LBHLHPuw3lJQhvDJY/view?usp=sharing

regards

closed 2022-05-13 by mpromonet