mpromonet/webrtc-streamer #410: When using Webrtc-Streamer, the HTML page of the foreground leaked the RTSP address. Is there any way to solve this problem?

UkVOSklF, 2021-04-16:
When using Webrtc-Streamer, the HTML page of the foreground leaked the RTSP address. Is there any way to solve this problem?

Is it possible to use encryption or other methods in the HTML page to prevent the RTSP address from being exposed in the page

Looking forward to your reply,thank you

mpromonet, 2021-04-16:
Hi @UkVOSklF,

The rtsp is show in the web browser if you start with :
```
webrtc-streamer <rtsp url>
```
But not if you start with : 
```
webrtc-streamer -nmycam -u <rtsp url>
```
In the live demo https://webrtc-streamer.herokuapp.com/, you don't see the rtsp url from the foreground.

Best Regards,
Michel.

UkVOSklF, 2021-04-16:
If I use it in a Windows environment, how do I do it?
Thanks for your advice

mpromonet, 2021-04-16:
Hi,

The same than on linux, using the command line or defining url in a config file.

Best Regards,
Michel.

closed 2021-04-16 by mpromonet