mpromonet/webrtc-streamer #413: when I tried to watch two rtsp stream,it happens time delay

JokerSora, 2021-04-22:
what caused the delay,how can I do with the problem

JokerSora, 2021-04-22:
er ,now some rtsp stream may happend time delay,so what makes it delay?the net work error a while?or the getIceCandidate api just delayed as  the chrome has a limit of request at one time. why not use websocket to realize the request with no block?

mpromonet, 2021-04-23:
Hi @JokerSora 

There is no problem to use websocket for signaling, as there is no emerging standard to do this (XMPP, different websocket implementation, ....), this project stay with the simplest signaling using ajax.
However if you like to implement websocket signaling, civetweb support it, so it needs some technical code to make the connection.

Best Regards,
Michel.

closed 2021-04-25 by JokerSora