mpromonet/webrtc-streamer #416: Desktop Capturer leaves impression after the menu is closed

prncvrm, 2021-05-13:
**Describe the bug**
Using desktop capturer in mac, leaves a impression when menu is closed. attached screenshot. The impression is majorly visible on a black background

**To Reproduce**
Steps to reproduce the behavior:
NA

**Expected behavior**
Shouldn't leave impressions

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Mac OS
 - Browser any
 - Version any

@mpromonet any help or suggestions you could give?

prncvrm, 2021-05-17:
@mpromonet any help/update?

mpromonet, 2021-05-17:
Hi @prncvrm 

I did not see this problem under Linux or Windows.
You may try https://www.webrtc-experiment.com/screen-sharing to discriminate the origin.

Best Regards,
Michel.

prncvrm, 2021-05-17:
Hi @mpromonet , 
thanks for the revert. couldn't understand the term `origin` ? can you elaborate please?

mpromonet, 2021-05-20:
Hi,

I was suggesting to use an other webrtc application that capture desktop.
An other idea could be to look at the code of `webrtc::DesktopCapturer`  and its options `webrtc::DesktopCaptureOptions::CreateDefault()`, it may have option to configure capturer you like to set ? https://github.com/mpromonet/webrtc-streamer/blob/master/inc/screencapturer.h#L18

Best Regards,
Michel.

prncvrm, 2021-05-20:
Hi,

I did set set options. added options like 
```
options.set_disable_effects(false);
options.set_allow_iosurface(true);
```
which i could find on https://source.chromium.org/chromium/chromium/src/+/main:content/public/browser/desktop_capture.cc;l=35?q=set_allow_iosurface&ss=chromium. Still the same.

prncvrm, 2021-05-20:
this is resolved, thanks

closed 2021-05-20 by prncvrm