mpromonet/webrtc-streamer #418: I found the release for window cannot be use by c# process to create a new process

JokerSora, 2021-05-25:
here is my code 
`Process p = new Process();
            p.StartInfo.FileName = @"D:\servers\webrtc-streamer\webrtc-streamer.exe";
            p.StartInfo.Arguments = "-H localhost:8000"; //+ args[0];//Parameters;
            p.StartInfo.UseShellExecute = false;
            p.StartInfo.RedirectStandardError = true;
            p.StartInfo.CreateNoWindow = false;
            p.StartInfo.RedirectStandardOutput = true;
            p.Start();`
howerver,if i cd D:\servers\webrtc-streamer\ and execute webrtc-streamer.exe -H localhost:8000 it works
then i found it went wrong if i use  D:\servers\webrtc-streamer\webrtc-streamer.exe -H localhost:8000 
the error info is here
[000:000][36004] (audio_device_generic.cc:28): BuiltInAGCIsAvailable: Not supported on this platform
[000:000][36004] (audio_device_generic.cc:38): BuiltInNSIsAvailable: Not supported on this platform

JokerSora, 2021-05-25:
even if I set the path to the system environment,it no works

mpromonet, 2021-05-25:
Hi,

You should look to the error code you get. You can also run with more verbosity to understand what makes exit.

Best Regards,
Michel.

JokerSora, 2021-05-26:
> Hi,
> 
> You should look to the error code you get. You can also run with more verbosity to understand what makes exit.
> 
> Best Regards,
> Michel.

en,thanks for your advise,howerver,I'm not familiar with c++,I'm a C# and javascript engineer,so here I use another way(using bat and copy the exe file for different processname) to realize the process manager on windows.

mpromonet, 2021-05-26:
Hi,

Your problem doesnot seems link to c++ langague, it is a process running with some arguments and that give a return code. If there is a popup window with a crash then a debugger may help you.

Best Regards,
Michel.

JokerSora, 2021-05-26:
> Hi,
> 
> Your problem doesnot seems link to c++ language, it is a process running with some arguments and that give a return code. If there is a popup window with a crash then a debugger may help you.
> 
> Best Regards,
> Michel.

thanks,I retried (using c# process) and found that the error had no crash popup window,the port was just token by the webrtc-streamer.exe,but when I open the ip:port,it just returned 404

mpromonet, 2021-05-26:
Hi,

You may use argument `-w <webroot>` to give path to look for html.

Best Regards,
Michel.

JokerSora, 2021-05-28:
> Hi,
> 
> You may use argument `-w <webroot>` to give path to look for html.
> 
> Best Regards,
> Michel.

thanks,I found it's no matter with the html file dictionary.even with it,the program works well.

closed 2021-05-28 by JokerSora