mpromonet/webrtc-streamer #419: what goes wrong,when I tried to transport multi rtsp to webrtc,the program stopped?

JokerSora, 2021-05-28:
what limited the muti rtsp to webrtc?
I use a server(server2012) with sixteen core and 64G memory, the net is 1Gbps
then I try to transport multi rtsp(each rtsp needs net about 15Mb/s)
when I tried more than three rtsp streams, the videoes in explorer were very terrible,then the program on server went wrong

I guess it's the problem of server computer's IO? or the live555 decode muti rtsp went wrong?

I need you advise,thank you very much!

mpromonet, 2021-05-28:
Hi,

Did you tried to use the `-o` option ? It is quite experimental but allow to not decode & encode, forwarding frame from rtsp to webrtc.

Best Regards,
Michel.

JokerSora, 2021-06-03:
> Hi,
> 
> Did you tried to use the `-o` option ? It is quite experimental but allow to not decode & encode, forwarding frame from rtsp to webrtc.
> 
> Best Regards,
> Michel.

It works! thanks very much!

closed 2021-06-03 by JokerSora