mpromonet/webrtc-streamer #424: FireFox WebRTC: ICE failed, add a TURN server and see about:webrtc for more details

andrewliu1345, 2021-06-11:
Use docker 
docker run  -p1985:1985 -p1985:1985/udp -p3478:3478/udp -p3479:3479 -p3479:3479/udp --restart=always --name="WebrtcServer" docker.io/mpromonet/webrtc-streamer -o -H1985 -S0.0.0.0:3478 -s192.168.0.193:3478

FireFox v91  WebRTC: ICE failed, add a TURN server and see about:webrtc for more details

[5634:470][60] (webrtc_sdp.cc:419): Failed to parse: "". Reason: Expect line: candidate:<candidate-str>
[5634:472][12] (webrtc_sdp.cc:419): Failed to parse: "". Reason: Expect line: candidate:<candidate-str>
[5634:565][23] (webrtc_sdp.cc:419): Failed to parse: "". Reason: Expect line: candidate:<candidate-str>
[5634:581][24] (webrtc_sdp.cc:419): Failed to parse: "". Reason: Expect line: candidate:<candidate-str>
[5634:583][10] (port.cc:380): Port[a4102470:0:1:0:local:Net[eth0:172.17.0.x/16:Ethernet:id=1]]: Received non-STUN packet from unknown address: 172.17.0.x:3478
[5634:586][10] (port.cc:380): Port[a4109140:0:1:0:local:Net[eth0:172.17.0.x/16:Ethernet:id=1]]: Received non-STUN packet from unknown address: 172.17.0.x:3478
[5635:083][10] (port.cc:380): Port[a4102470:0:1:0:local:Net[eth0:172.17.0.x/16:Ethernet:id=1]]: Received non-STUN packet from unknown address: 172.17.0.x:3478
[5635:086][10] (port.cc:380): Port[a4109140:0:1:0:local:Net[eth0:172.17.0.x/16:Ethernet:id=1]]: Received non-STUN packet from unknown address: 172.17.0.x:3478
[5636:083][10] (port.cc:380): Port[a4102470:0:1:0:local:Net[eth0:172.17.0.x/16:Ethernet:id=1]]: Received non-STUN packet from unknown address: 172.17.0.x:3478
[5636:086][10] (port.cc:380): Port[a4109140:0:1:0:local:Net[eth0:172.17.0.x/16:Ethernet:id=1]]: Received non-STUN packet from unknown address: 172.17.0.x:3478
[5638:083][10] (port.cc:380): Port[a4102470:0:1:0:local:Net[eth0:172.17.0.x/16:Ethernet:id=1]]: Received non-STUN packet from unknown address: 172.17.0.x:3478
[5638:087][10] (port.cc:380): Port[a4109140:0:1:0:local:Net[eth0:172.17.0.x/16:Ethernet:id=1]]: Received non-STUN packet from unknown address: 172.17.0.x:3478
[5642:083][10] (port.cc:380): Port[a4102470:0:1:0:local:Net[eth0:172.17.0.x/16:Ethernet:id=1]]: Received non-STUN packet from unknown address: 172.17.0.x:3478
[5642:087][10] (port.cc:380): Port[a4109140:0:1:0:local:Net[eth0:172.17.0.x/16:Ethernet:id=1]]: Received non-STUN packet from unknown address: 172.17.0.x:3478
[5650:083][10] (port.cc:380): Port[a4102470:0:1:0:local:Net[eth0:172.17.0.x/16:Ethernet:id=1]]: Received non-STUN packet from unknown address: 172.17.0.x:3478
[5650:087][10] (port.cc:380): Port[a4109140:0:1:0:local:Net[eth0:172.17.0.x/16:Ethernet:id=1]]: Received non-STUN packet from unknown address: 172.17.0.x:3478
[5658:083][10] (port.cc:380): Port[a4102470:0:1:0:local:Net[eth0:172.17.0.x/16:Ethernet:id=1]]: Received non-STUN packet from unknown address: 172.17.0.x:3478
[5658:087][10] (port.cc:380): Port[a4109140:0:1:0:local:Net[eth0:172.17.0.x/16:Ethernet:id=1]]: Received non-STUN packet from unknown address: 172.17.0.x:3478
[5666:083][10] (port.cc:380): Port[a4102470:0:1:0:local:Net[eth0:172.17.0.x/16:Ethernet:id=1]]: Received non-STUN packet from unknown address: 172.17.0.x:3478
[5666:087][10] (port.cc:380): Port[a4109140:0:1:0:local:Net[eth0:172.17.0.x/16:Ethernet:id=1]]: Received non-STUN packet from unknown address: 172.17.0.x:3478

mpromonet, 2021-06-11:
Hi,

You may probably need to improve your issue description.

Best Regards,
Michel.

andrewliu1345, 2021-06-15:
> Hi,
> 
> You may probably need to improve your issue description.
> 
> Best Regards,
> Michel.

So sorry!
Thank you for your reply. 
I'll revise it right away.
Thanks~~!

closed 2022-05-13 by mpromonet