mpromonet/webrtc-streamer #425: Adding Audio sending capability from web page to Rpi server

erwan2345, 2021-06-22:
Hello, 
First, I wanted to congratulate you for the job about V4L2 (loopback, copy etc..)  and also with this webrtc-streamer.

My webrtc-streamer environment:
   I could make it working, from a chrome page (either from a windows laptop or an iphone in SSL) I could open a video stream coming from my RPi and getting sound from it (thks to a USB microphone).

   I could use directly the raspicam in H264 and also, using the v4l2copy utility,  a virtual device (I have 2 virtual device so that I can use the raspicam video in several applications at the same time).

My target is to make a video doorbell able to:
   - permanently send a RTSP H264 + Audio  stream to my NAS  (ok:  thks to 4l2copy and v4l2rtspserver)
   - on demand: provide a webrtc communication to mobile phone web page (IOS & android),  audio 2 way, video from RPi to webpage from outside local network.

My request:
   Webrtc-streamer allow 1-way communication sending video+audio from RPI to any webpage.

   Would it be possible to add a reverse communication  (audio only is sufficient for my use) using webrtc  from webpage to rpi   audio jack out ? I'm not a big specialist, but please advise me in which area the code should be updated ? 

Thks

mpromonet, 2021-06-22:
Hi,

WebRTC is 2-way connection, you can look to https://github.com/mpromonet/webrtc-streamer-html/blob/master/sendstream.html that was a try to send a stream from web browser to webrtc-streamer. The stream is notified in [PeerConnectionObserver ::OnAddStream](https://github.com/mpromonet/webrtc-streamer/blob/master/inc/PeerConnectionManager.h#L213). The video sink is just printing frames size, and audio sink is ignored, but you can send it to a audio device if you like.

Best Regards,
Michel.

erwan2345, 2021-06-22:
Thks for the quick reply, I was wondering if it was requiring an update in the server deep code. Here I understand that I can simply update the html page with the right request. I will the sample you provided, I’ll try to update from video to audio and will post an update.
  
  
  
  
 Bye
 Erwan

mpromonet, 2021-06-22:
Hi Erwan,

It needs to update the server side code to get the AudioSink and implement it.

Best Regards,
Michel.

erwan2345, 2021-06-23:
Hello Michel
  

  
I entered into the code but I’m afraid that my low coding skills will not generate a working result, I’ll keep trying but it may take time, then if you or other skilled people are able to have a look, I’m sure that a lot of other people are dreaming about this audio back addition in webrtc-streamer which is just working very well on video point of view.
  

  
Thks   
  
  
 Erwan Olive

mpromonet, 2021-06-23:
Hi Erwan,

Don't be afraid, this is a really small project, the tricky part is probably to manage multiple session, and choose how to output the audio (using WebRTC SDK or not). If you are starting coding in C++, it may be a good opportunity to dive a learn a bit more.

Best Regards,
Michel.

AleXoundOS, 2021-08-23:
@erwan2345, have you succeded?