mpromonet/webrtc-streamer #427: Cannot See Video in Janus Video Room.

Meonardo, 2021-06-28:
Thanks for the great repo!
I have a problem publishing RTSP stream to my local Janus Video Room,
here are the steps what I am doing:
1. Run WebRTC-Streamer by using Docker image(macOS, x86):
`docker run -p 8000:8000 -it mpromonet/webrtc-streamer -vvv -n LocalCamera201 -u rtsp://192.168.5.201:554/main.h264 -sstun.l.google.com:19302`
2. Configure Janus Video Room:
`room-1234: {
       ......
        videocodec = "h264"  # set to vp8 has the same issue.
......
}`
3. I can see my local IP camera `(rtsp://192.168.5.201:554/main.h264`) content at `localhost:8000`, 
when I switch to `localhost:8000/janusvideoroom.html` select it and publish to JanusVideoRoom, nothing appears,
but it does join the videoroom(just no video shows).

---
Please see the log files below:

[Janus.log](https://github.com/mpromonet/webrtc-streamer/files/6724032/Janus.log)
[Streamer.log](https://github.com/mpromonet/webrtc-streamer/files/6724033/WebRTC-Streamer.log)

Can you give some advice?

mpromonet, 2021-06-28:
Hi Meodarno,

The first thing that come in my mind, is to try adding `--net host` (or using the executable outside docker). There is something that I still not understood with ICE candidate enumeration in a docker, it may give internal docker IP that are usualy not accessible from outside.

Best Regards,
Michel.

Meonardo, 2021-06-29:
Thanks for your reply!
> adding `--net host`.

Sadly, Its not supported by Docker Desktop for macOS(according to this [link](https://docs.docker.com/network/host/)),
Then I switch to Ubuntu 16.04, I dont know if its a codec issue or not(tried the example RTSP stream and my local IP camera stream) the JanusVideoRoom give me an error of `WebRTC error... Failed to set remote offer sdp: Failed to set remote video description send parameters.`

here is the log: 
[log.log](https://github.com/mpromonet/webrtc-streamer/files/6730390/log.log)

> using the executable outside docker

I've tried to execute in Windows use the latest relase version and it gives me the issue as I described above.

> it should be available from "Video Room" of janus.

You're right, my mistake.

Again, thanks for your time to give a reply!

closed 2021-11-01 by mpromonet