mpromonet/webrtc-streamer #428: Cannot run webrtc-streamer on windows 10

pengcx123, 2021-07-12:
**Describe the bug**
I download the web-streamer for windows, and use below command to run web-streamer:
webrtc-streamer rtsp://localhost/32050000001120000569_0500000570
It exit with error
**To Reproduce**
Steps to reproduce the behavior:
1. run webrtc-streamer rtsp://localhost/32050000001120000569_0500000570

**Expected behavior**
It can start and run ok

**Screenshots**
D:\temp\webrtc-streamer-v0.6.4-Windows-AMD64-Release\webrtc-streamer-v0.6.4-Windows-AMD64-Release>webrtc-streamer rtsp://localhost/32050000001120000569_0500000570
Version:v0.6.4/Windows-AMD64 civetweb@v1.13 webrtc@82aa094a97-dirty live555helper@d351689
{
        "urls" :
        {
                "rtsp://localhost/32050000001120000569_0500000570" :
                {
                        "video" : "rtsp://localhost/32050000001120000569_0500000570"
                }
        }
}Logger level:3
[000:000][18008] (audio_device_generic.cc:28): BuiltInAGCIsAvailable: Not supported on this platform
[000:000][18008] (audio_device_generic.cc:38): BuiltInNSIsAvailable: Not supported on this platform
HTTP Listen at 0.0.0.0:8000
cannot bind to 0.0.0.0:8000: 10048 (No error)
Failed to setup server ports
Cannot Initialize start HTTP server exception:null context when constructing CivetServer. Possible problem binding to port.
Exit

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Windows 10
 - Browser [e.g. chrome, safari] Chrome
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

pengcx123, 2021-07-12:
We will setup the application on Linux

closed 2021-07-12 by pengcx123

mpromonet, 2021-07-12:
Hi pengcx123,

It is surely a better way to use Linux, it build on windows, but some features are not implmented in windows.
By the way it seems it fails to listen on port 8000. 
Commonly on windows a popup is openned to ask persmission to bind port. 

Best Regards,
Michel.

pengcx123, 2021-07-12:
Thanks for your quick reply, it is very helpful.
I closed this issue and will test on linux later.

Send time: Monday, Jul 12, 2021 5:41 PM