mpromonet/webrtc-streamer #429: how to install in centos?  I run the docker, but it doesn't work.

zhuchuanyue, 2021-07-12:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

mpromonet, 2021-07-12:
Hi zhuchuanyue,

In order to receive help, you need to describe what means "doesn't work".
Maybe some error messages ?

Best Regards,
Michel.

zhuchuanyue, 2021-07-12:
[image]

mpromonet, 2021-07-12:
Hi zhuchuanyue,

Then it is running and receive calls. If you don't have video, it maybe because of ICE negociation in docker. Adding `--net=host` may help.

Best Regards,
Michel.

zhuchuanyue, 2021-07-13:
ok，thanks, I can have the  video.

zhuchuanyue, 2021-07-13:
[image]

Hi, I open one live  video,  the cpu is over 100%,  how  can  I   resolve  this   question ?

mpromonet, 2021-07-13:
Hi zhuchuanyue,

An option to reduce the cpu consomption is to use the `-o` argument, it works only for H264 source and will forward the frame. Then there is no possible adaptation (codec, scale, gop, ....) but it use less cpu removing decoding/encoding.

Best Regards,
Michel.

zhuchuanyue, 2021-07-14:
yes, I try it  by use the -o,  the cpu reduced,  but  the video loading is slow， about  10 seconds.

mpromonet, 2021-07-17:
Hi,

This depends on the gop of your backend, the first image need to wait for an I-frame, appending more I-frame to the backend stream should reduce this.

Best Regards,
Michel.

szueric, 2021-07-24:
遇到跨域的问题，怎么处理，请高手帮忙？

```
Access to fetch at 'http://192.168.12.19:18010/api/call?peerid=0.8798291299329237&url=rtsp%3A%2F%2F192.168.12.231%3A8557%2Fh264&options=rtptransport%3Dtcp%26timeout%3D60' from origin 'http://127.0.0.1:5500' has been blocked by CORS policy: No 'Access-Control-Allow-Origin' header is present on the requested resource. If an opaque response serves your needs, set the request's mode to 'no-cors' to fetch the resource with CORS disabled.
```

lizhen789, 2022-06-24:
windows11 下
[image]

mpromonet, 2022-06-24:
Hi @lizhen789 

May you read the issue before posting something that seems completly different issue.... windows and centos seems different os ?

Thanks,
Michel.

closed 2022-08-15 by mpromonet