mpromonet/webrtc-streamer #430: Cannot share video on Jitsi meet

pengcx123, 2021-07-12:
**Describe the bug**
Jitsitest.html cannot share the video on jitsi meet
**To Reproduce**
Steps to reproduce the behavior:
1. Start webstreamer on our computer, its ip is 47.94.157.157
2. Start a jitsi meet on our jitsi meet systerm
3. Go to http://47.94.157.157:8000/jitsitest.html
4. An user attend the jitsi meet, without any video, and the camera was turned off

**Expected behavior**
The user's window will display the video

**Screenshots**
Bellow is the content of jitsitest.html
<html>
<head>
<script src="libs/strophe.min.js" ></script>
<script src="libs/strophe.muc.min.js" ></script>
<script src="libs/strophe.disco.min.js" ></script>
<script src="libs/strophe.jingle.sdp.js"></script>
<script src="libs/jquery-1.12.4.min.js"></script>
<script src="libs/EventEmitter.min.js" ></script>
<script src="xmppvideoroom.js" ></script>
<script src="xmppvideoroom.json" ></script>
<script src="webrtcstreamer.json" ></script>
</head>
<body></body>
<script>   
	function writeStatus(name, status) {
		var textNode = document.createElement("h3");
		textNode.innerHTML = name + " " + status;
		document.body.appendChild(textNode);
	}
	var bus = new EventEmitter();
	bus.addListener('state', writeStatus);
	var xmpp = new XMPPVideoRoom(xmppRoomConfig.url, webrtcConfig.url, bus);
	xmpp.join(xmppRoomConfig.roomId, "rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov","Bunny");
	
	window.onbeforeunload = function() { 
			xmpp.leave();
	};
</script>
</html>

Below is the configuration:
var xmppRoomConfig = {
	url: "videocon.01more.com",
	roomId: "test_123456"
}

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Ubuntu 18.04
 - Browser [e.g. chrome, safari] Chrome
 - Version [e.g. 22] 91.0.4472.124

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

pengcx123, 2021-07-12:
The webstreamer version is v0.5.2

pengcx123, 2021-07-12:
I debug the chrome, found below error
xmppvideoroom.js:278 Fetch API cannot load rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov/api/call?peerid=3mbvlvuvbvdqe&url=rtsp%3A%2F%2Fwowzaec2demo.streamlock.net%2Fvod%2Fmp4%3ABigBuckBunny_115k.mov&options=rtptransport%3Dtcp%26timeout%3D60. URL scheme must be "http" or "https" for CORS request.

mpromonet, 2021-07-12:
Hi pengcx123,

The jitsitest.html page is not supposed to display a video, it is supposed to join a jitsi server and publish a webrtc stream.
You should look to the jistsi meeting.
However, jitsi interface seems to evolve, it is highly possible that it needs some debug/fix. janus-gateway interface is more stable.

Best Regards,
Michel.

pengcx123, 2021-07-12:
Thanks for your information, in the jitsi meet page, I found the camera of webrtc user was turnned off and there was nothing in its window.
In the previous version of jitsi that work correctly, was there a video stream display in the window? or my understanding was totally wrong?

Send time: Monday, Jul 12, 2021 5:46 PM

mpromonet, 2021-07-12:
Hi pengcx123,

I made a quick test using webrtc-streamer on my PC and jit.si, and as you said the camera from the jitsi is disabled and no video is displayed, but video is displaied from xmppvideoroom.html. Maybe not a complex problem...

Best Regards,
Michel.

pengcx123, 2021-07-12:
Thanks a lot, I will test it

Send time: Monday, Jul 12, 2021 6:16 PM

pengcx123, 2021-07-13:
Thanks a lot for your help, and there is one more question, can this project support to bypass the webrtc server and directly share a WebRTC stream to Jitsi meet? this is for below case:
1)We have a RTSP stream source
2)We have an application which convert the RTSP stream to WebRTC stream
3)We want to share the WebRTC stream to a jitsi meet

Send time: Monday, Jul 12, 2021 6:17 PM

Subject:  Re:  [mpromonet/webrtc-streamer] Cannot share video on Jitsi meet (#430)

Send time: Monday, Jul 12, 2021 6:16 PM

pengcx123, 2021-07-13:
Thanks a lot for your help,  in my test, I coud see the video window, but I could not see the content of the video.
The details are in the two attached files.
Are there any plugins needed to install in the Chrome?
Would you mind to have a look?

Best Regards,
Bert Peng.

Send time: Monday, Jul 12, 2021 6:16 PM

pengcx123, 2021-07-13:
Hi Michel,
After more test, I succeeded to display the video in XMPPVideoRoom page, thanks for your help a lot.
I am tring to bring the video to the meet window now.

Best Regards,
Bert Peng

Send time: Monday, Jul 12, 2021 6:16 PM

pengcx123, 2021-07-14:
Hi Michel,
Thanks for your help. 
I did more test, and would like to ask below questions:
1) I tested on several computer, I openned a chrome and hosted a jitis meeting in it, and then openned xmpproom.html page, the computers were able to share the video in xmpproom.html page, but were not able to see them in the jitsi meeting, so I wondered whether was it the legacy behavor or introduced change?
2) I tested the webrtc-streamer, changed the rtsp source to my own source, which was rtsp://39.105.135.70/34020000001110000070_0200000001, but I got below error:

[h264 @ 0x7effc4016400] no frame!
[053:818][20108] (h264_decoder_impl.cc:280): avcodec_send_packet error: -1094995529
[053:818][20108] (VideoDecoder.h:70): VideoDecoder::DecoderThread failure:-1
[h264 @ 0x7effc4016400] non-existing PPS 0 referenced
[h264 @ 0x7effc4016400] decode_slice_header error

Would you mind to give me some suggestion?

Best Regards,
Bert Peng

Send time: Tuesday, Jul 13, 2021 2:34 PM

Subject:  Re:   [mpromonet/webrtc-streamer] Cannot share video on Jitsi meet (#430)

Send time: Tuesday, Jul 13, 2021 12:04 PM

Subject:  Re:  [mpromonet/webrtc-streamer] Cannot share video on Jitsi meet (#430)

Thanks a lot for your help,  in my test, I coud see the video window, but I could not see the content of the video.

Send time: Monday, Jul 12, 2021 6:16 PM

pengcx123, 2021-07-16:
Maybe it is a request, not an issue

closed 2021-07-16 by pengcx123

zsinba, 2021-12-29:
> 

what about your issue