mpromonet/webrtc-streamer #431: cant handle rtsp address

liyoubdu, 2021-07-14:
webrtc-streamer  can not handle the rtsp format below:
`./webrtc-streamer rtsp://192.168.1.5:12345/live?skip&video=1&audio=1&intervieweeID=263`

commit: 4922c95934b084a0f97a239d838dd36969ee1f99

mpromonet, 2021-07-14:
Hi,

I guess you have a problem in the way you pass the parameter.
Did you tried  like this ?

`./webrtc-streamer "rtsp://192.168.1.5:12345/live?skip&video=1&audio=1&intervieweeID=263"`

Best Regards,
Michel.

liyoubdu, 2021-07-14:
> Hi,
> 
> I guess you have a problem in the way you pass the parameter.
> Did you tried like this ?
> 
> `./webrtc-streamer "rtsp://192.168.1.5:12345/live?skip&video=1&audio=1&intervieweeID=263"`
> 
> Best Regards,
> Michel.

it works. thank you

liyoubdu, 2021-07-14:
use ./webrtc-streamer "rtsp://192.168.1.5:12345/live?skip&video=1&audio=1&intervieweeID=263". and it works.

closed 2021-07-14 by liyoubdu