mpromonet/webrtc-streamer #432: Can't build for armv6l

imerso, 2021-07-15:
In Dockerfile.rpi it is specified that the build should be for armv6l, but the resulting binary is for armv7. The pre-built webrtc-streamer runs perfectly, but I have been changing the code for my own needs, so I need to rebuild it. Rebuilding always generates armv7, which runs ok on pi2 and above, but I need it running on pi zero armv6.

I noticed that webrtc itself install_sysroot will install armv7 binaries when arm build is selected.

Is there anything we can do about that? Do you have, and could possibly share a compatible armv6 sysroot? Is the file debian_jessie_arm_sysroot that is needed?

Thank you!

mpromonet, 2021-07-16:
Hi,

By default (without build-arg) it should build armv6 binary (that run should run on zero). You may look to [.cirrus.yml](.cirrus.yml) that build image abailable from dockerhub.

Best Regards
Michel

imerso, 2021-07-18:
Thank you very much for the attention! In the end, it was the cross-compiler version. Shame on me, I was using my pre-installed cross-compiler, but it was insisting either on armv7 or a too new glibc. I took the one you have on your dockerfile, and it is now working properly! Nice!

closed 2021-07-18 by imerso