mpromonet/webrtc-streamer #433: Getting [h264 @ 0x7fdbfc010240] no frame!

subhadeep-123, 2021-07-16:
Buiding webrtc from docker

[image]

Tried adding a patch from on of the closed issue reported back in 2018 - (Link to the issue - https://github.com/mpromonet/webrtc-streamer/issues/137)

Here is our modified dockerfile

```
FROM ubuntu:20.04 as builder
WORKDIR /webrtc-streamer
COPY VideoDecoder.patch /d
RUN apt-get update && DEBIAN_FRONTEND=noninteractive apt-get install -y --no-install-recommends ca-certificates wget git python python-pkg-resources python3 python3-pkg-resources g++ autoconf automake libtool xz-utils libpulse-dev libasound2-dev libgtk-3-dev libxtst-dev libssl-dev cmake make pkg-config p7zip-full patch \
    && git clone https://chromium.googlesource.com/chromium/tools/depot_tools.git /webrtc/depot_tools \
    && export PATH=/webrtc/depot_tools:$PATH \
    && git clone https://github.com/mpromonet/webrtc-streamer.git . \
    && cp /VideoDecoder.patch /webrtc-streamer/inc/ \
    && cd /webrtc-streamer/inc/ \
    && patch < VideoDecoder.patch \
    && cd /webrtc \
    && fetch --no-history --nohooks webrtc \
    && sed -i -e "s|'src/resources'],|'src/resources'],'condition':'rtc_include_tests==true',|" src/DEPS \
    && gclient sync \
    && cd /webrtc-streamer \
    && cmake . && make \
    && cpack \
    && mkdir /app && tar xvzf webrtc-streamer*.tar.gz --strip=1 -C /app/ \
    && rm -rf /webrtc && rm -f *.a && rm -f src/*.o \
    && apt-get clean && rm -rf /var/lib/apt/lists/
FROM ubuntu:20.04
WORKDIR /app
COPY --from=builder /app/ /app/
RUN apt-get update && DEBIAN_FRONTEND=noninteractive apt-get install -y --no-install-recommends libssl-dev libasound2 libgtk-3-0 libxtst6 libpulse0 avahi-utils jq wait-for-it \
    && apt-get clean && rm -rf /var/lib/apt/lists/ \
    && ./webrtc-streamer -V
COPY rootfs /
RUN chmod +x /start.sh
ENTRYPOINT [ "/start.sh" ]
```

Tested in both mac and windows. Point to note here - **the stream is 1 FPS, and over TCP**

mpromonet, 2021-07-16:
Hi,

Taking code from a 2 years old discussion may be no more appropriated.

Best Regards
Michel

subhadeep-123, 2021-07-19:
@mpromonet Thanks for your response, I think the situation is not a problem.

closed 2021-07-19 by subhadeep-123